realsenseai/librealsense #3852: Segmentation F

filipematosinov, 2019-04-26:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 } | 
| Firmware Version                   | (Open RealSense Viewer -->  2.20) | 
| Operating System & Version |   {Ubuntu Mate 18.04.2}  | 
| Kernel Version (Linux Only)    |  (4.15.0-1033-raspi2)                                         | 
| Platform                                 | Raspberry Pi |
| SDK Version                            |  { 2.20 }                          | 
| Language                            |  {python}                          | 
| Segment			|  {others }                          | 

### Issue Description
I am  trying to use my Raspberry Pi 3 to work with the realsense cameras. However I am experienced several Segmentation faults using both the realsense-viewer as well as scripts in python. Sometimes I can get two cameras work at the same time without problems some other time I get segmentation fault after a while or just at start streaming. I have seen that there are other people with the same issue but no solution has been found

ev-mp, 2019-04-30:
@filipematosinov , kernel log, as well as Librealsense's log  would definitely help here.
Note that if the issue is most often occurs **some time after** the streaming starts, then I would first check the USB connection, specifically tried to switch to a powered USB3 hub to negate power-related/CPU throttling issues.

RealSenseCustomerSupport, 2019-05-30:
----------------------------------------------

Closing this issue due to lack of activity for 30+ days.

closed 2019-05-30 by RealSenseCustomerSupport