realsenseai/librealsense #3853: Multicam in Matlab

JimmyFroost, 2019-04-26:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 + T265 | 
| Firmware Version                   | (05.11.01.100) / (0.0.18.5448)  | 
| Operating System & Version |   {Win ( 10)    |  
| Platform                                 | PC  |
| SDK Version                            |  {   2.2}                          | 
| Language                            |Matlab 2018                     |  

Hello

I've read several issues about this but none  solved my problem :
I want to use T265 and D415 at the same time, but whene both are connected to my laptop, only D415 is detected. 
When I run the matlab program, I even hear the sound as if T265 was disconnected from usb port.

Another thing, this code only returns 1 when both are connected, 0 when T265 is connected.

```
ctx = realsense.context();
n_devices = length(ctx.query_devices())
```
I can identify them (running separately) by their SNumber but I'm not sure I can get anything  starting from that.

I thought [this issue ](https://github.com/IntelRealSense/librealsense/issues/3733) would help me but the code doesn't work for me...

thanks

MartyG-RealSense, 2019-04-26:
There was another recent case where T265 was not detected by MATLAB.   The user in that case solved the problem by putting a **clear** instruction at the start of their script.

https://github.com/IntelRealSense/librealsense/issues/3774

JimmyFroost, 2019-04-29:
Yes I've read that case already and I have 'clear' in my code.
I guess there is something wrong in T265 Matlab implementation by Intel. In #3733 evoqued, in my first post, the command **ctx.query_devices()** would work fine but it doesn't with this new camera...

JimmyFroost, 2019-05-13:
I just installed SDK 2.21, and this issue is unfortunately not solved. T265 is not detected if D415 is connected to the laptop. I always get :
 **length(ctx.query_devices()) = 0** when T265 is connected 
 **length(ctx.query_devices()) = 1** when both are connected 

One thing I managed is to initialize a pipe for T265, with D415 not yet connected to the laptop, then plug D415 and initialize a new pipe for it.
This way, I have access to each pipe, frameset and data eventually.

So I make it work but it's not a perfect operational mode at all, I can't keep unplugging and plugging camera D415 again and again.

closed 2020-02-04 by RealSenseCustomerSupport