realsenseai/librealsense #3854: Simple method Jetson nano support (via dpkg) [more kernel versions supported]

eric-schleicher, 2019-04-26:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435/  T265  | 
| Operating System & Version |   Ubuntu L4T Jetpack 4.2 32.1 | 
| Kernel Version (Linux Only)    |  kernel 4.9                                         | 
| Platform                                 | NVIDIA Jetson|
| SDK Version                            |  2.x (current) | 

### Issue Description
Availability of precompiled librealsense dpkgs do not [currently] overlap with mainline jetson releas(es) by vernel version.  e.g. Currently the GA JetPack 4.2 (L4T32.1) ships with kernel 4.9; but realsense build system does not produce compatible binaries.

This means that in order to get realsense working on Jetson Nano; a complicated. lengthy and error prone compilation (+/or cross compilation environments).  This is beyond the ability/interest of most users.  Despite the great resource that JetsonHacks and video are; they should not be relied upon as the sole method to get realsense cameras working on Jetson devices (disqualifying mention of this as a solution).

Given the popularity of the nano, it's warrants some directed effort to get it properly supported with RealSense camera.

It would be ideal to have Realsense SDK binaries available for more kernels versions given the effort/value to users.

mzahana, 2019-04-27:
This is becoming more required. What is the current solution to get D435/T265 working with the nano and utilize its GPU especially working ROS?

dorodnic, 2019-04-30:
@mzahana - currently for D435i you'd need to manually adjust and apply kernel patches.
For everything else - [libuvc_installation.md](https://github.com/IntelRealSense/librealsense/blob/master/doc/libuvc_installation.md) is the fastest route. 
@eric-schleicher - your request clearly makes sense. We will try to prioritize this.

mzahana, 2019-05-01:
@dorodnic thanks for the reply. Actually, I am not sure how to manually adjust the kernel to make it work with Jetson Nano. If you can share some resources on how to do that, will be very much appreciated. 

Thanks.

Mango-kid, 2019-05-01:
This might be a helpful reference. Not sure about support for the Nano. 

[https://github.com/jetsonhacks/buildLibrealsense2TX](url)

christopherbate, 2019-05-02:
@mzahana 

I posted some instructions for the Nano on the other Nano thread:

https://github.com/christopherbate/Nano-Realsense-Instructions

MartyG-RealSense, 2019-05-09:
@mzahana I just saw an interesting newly released YouTube video about using the Nano with T265.

https://www.youtube.com/watch?v=NonBlt-wKCI

mzahana, 2019-05-10:
Yes. I saw that. I was waiting for that!

atyshka, 2019-07-17:
Still interested in this. The instructions in this repo are currently outdated for newer Jetpack releases and devices. Jetsonhacks tries to maintain a fair amount of resources but they can't keep up with all the latest software releases. arm64 packages and dkms would really help a lot, especially considering the realsense family is perhaps the number one device used with Jetson devices

dorodnic, 2019-08-04:
@eric-schleicher @atyshka @mzahana @christopherbate 

Update on our activities to support this requirement: 

a. Firmware update tool, previously available for x86 only was open-sourced and is now available together with the SDK for Jetson
b. SR300, D415, D435 and T265 cameras can be used on Jetson via [regular Debian installation instructions](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md)

There is still work to be done - D435i requires kernel patch and build from source, pip packages are not yet available, calibration stack is not available. Will continue to update.

GerardMaggiolino, 2019-08-28:
@dorodnic Just to clarify, are you stating that SR300, D415, D435 and T265 cameras should be compatible with the Jetson Nano per that linked guide? 

Attempting to follow that guide on the Jetson Nano yields the following error: 
```
N: Skipping acquire of configured file 'main/binary-arm64/Packages' as repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic InRelease' doesn't support architecture 'arm64'
```
which, when googling around, leads me to #1049 where the recommended method is using kernel patches.

rishabhkjain, 2019-09-16:
@samxchandlerpi I had the same issue when trying to follow that guide. I just followed this guide and it worked with my D435. https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

samxchandlerpi, 2019-09-16:
> @samxchandlerpi I had the same issue when trying to follow that guide. I just followed this guide and it worked with my D435. https://github.com/IntelRealSense/librealsense/blob/master/doc/installation.md

Thanks for the reference. I ended following this guide to get the d435 working:

https://www.jetsonhacks.com/2019/05/16/jetson-nano-realsense-depth-camera/

nadav-anduril, 2019-11-21:
> @eric-schleicher @atyshka @mzahana @christopherbate
>
> Update on our activities to support this requirement:
>
> a. Firmware update tool, previously available for x86 only was open-sourced and is now [...]

@dorodnic , can we expect dynamic calibration api to be available for jetson in the near future?

dorodnic, 2019-12-10:
@nadav-anduril - I cannot provide schedule for dynamic calibration availability (it is separate and fairly complicated software package) but we are integrating streamlined calibration capabilities into the SDK. White paper is coming, and these are fully available on all platforms.

closed 2020-02-18 by dorodnic