realsenseai/librealsense #3857: Firmware Device Version Issue

barnjamin, 2019-04-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.6.200 | 
| Operating System & Version |  ubuntu 18.04  | 
| Kernel Version (Linux Only)    | 4.15.0-47  | 
| Platform                                 | PC  |
| SDK Version                            |  bec4d4a8cd1056ddbd42fd0fa128b61330885382  | 
| Language                            |  N/A   | 

Just updated firmware on new D435 device and it now thinks its a D430 so no RGB Camera is available.

```
Device info: 
    Name                          : 	Intel RealSense D430
    Serial Number                 : 	838212071530
    Firmware Version              : 	05.11.06.200
    Recommended Firmware Version  : 	05.10.03.00
    Physical Port                 : 	/sys/devices/pci0000:00/0000:00:14.0/usb2/2-2/2-2:1.0/video4linux/video0
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0AD4
    Usb Type Descriptor           : 	3.2
```

MartyG-RealSense, 2019-04-27:
This problem has happened to other users in a small number of cases.  One remedy to try is to downgrade to an older firmware and then re-upgrade to the newer version.

Please bear in mind that the firmware only allows 20 downgrades before it locks the ability to downgrade and you can then only upgrade to newer versions than the one currently installed.  So if this does not work for you the first time, I recommend coming back here for further advice.  Thanks!

barnjamin, 2019-04-28:
This worked, thanks!

closed 2019-04-28 by barnjamin

MartyG-RealSense, 2019-04-28:
Awesome news - glad I could help!   :)

zjcs, 2019-05-24:
Same problem happen to me. 
I try to downgrade to V5.11.4 or V5.10.13, then upgrade to V5.11.6.200. But I fail to find D435i back.
After I read this issue, I choose the minimal FW version V5.10.3 showed on RSViewer for my D435i, then upgrade to V5.11.6.200. 

Done.