realsenseai/librealsense #3859: What is the difference between the .ply file generated by intelRealSense and Kinect?

bingws, 2019-04-28:
I have two .ply files : one generated by intelRealSense, as shown in figure 1. One was generated by Kinect, as shown in figure 2 . How do we switch?
[image]
figure 1

[image]
figure 2

dorodnic, 2019-04-28:
The PLY generated by RealSense is in binary format. It is valid, and can be opened using MeshLab application.

bingws, 2019-04-28:
Can I export with a python program?

bingws, 2019-04-29:
I really need this feature. Can anyone help me? thanks.

MartyG-RealSense, 2019-04-29:
If you want to view a ply generated with RealSense, you could either use MeshLab as Dorodnic suggested, or try loading it into a Python-based point cloud program called Pyntcloud.

https://github.com/daavoo/pyntcloud

bingws, 2019-05-05:
[image]
I tried to use Pyntcloud to transform the binaries bingws.ply to ascii bingws1.ply.There was an error.

RealSenseCustomerSupport, 2019-05-06:
----------------------------------------------

In addition to the suggestion from MartyG-RealSense, there is also a python module to parse PLY file.
https://github.com/dranjan/python-plyfile

closed 2019-05-08 by bingws