realsenseai/librealsense #3861: Firmware

lykGH, 2019-04-28:
after upgrading the firmware,the computer can not recognize D415

lykGH, 2019-04-28:
[图片]

ev-mp, 2019-05-01:
@lykGH, please provide a detailed description:
- Environment, OS, Platform
- Old/New Firmware versions. DFU Tool/version used
- SDK version, SDK demos/tools that you've tried.
- Kernel/Librealsense/Console/Device manager logs

lykGH, 2019-05-02:
My computer environment is window 10.The firmware version is D415 and the SDK is SDK2.0Before the hardware updated, the computer can use the device.However, when it reminds me to update, i accidentally disconnected the connection with the computer, and then the computer can not recognize the device.What should I do?-------- 原始邮件 --------发件人： ev-mp <notifications@github.com>日期： 昨天下午4:06收件人： IntelRealSense/librealsense <librealsense@noreply.github.com>抄送： lykGH <846296206@qq.com>, Mention <mention@noreply.github.com>主    题： Re: [IntelRealSense/librealsense] Firmware (#3861)@lykGH, please provide a detailed description:
Environment, OS, PlatformOld/New Firmware versions. DFU Tool/version usedSDK version, SDK demos/tools that you've tried.Kernel/Librealsense/Console/Device manager logs

ev-mp, 2019-05-02:
@lykGH , can you attach screenshots of the Device Manager with the device attached, as described in #3785 ? Also  - what FW version did use to upgrade the device ?
You can also run `rs-enumerate-devices` tool found in the Windows installer and attach the log.

lykGH, 2019-05-03:
uwp Drivers v6.1Before the update，device identification appears under the camera，It can be used normally.But now it is identified in another place，nothing about the program can be run. This program （rs-enumerate-devices )also cannot run,because this program will flash back.-------- 原始邮件 --------发件人： ev-mp <notifications@github.com>日期： 昨天半夜11:52收件人： IntelRealSense/librealsense <librealsense@noreply.github.com>抄送： lykGH <846296206@qq.com>, Mention <mention@noreply.github.com>主    题： Re: [IntelRealSense/librealsense] D415 is not recognized after firmware update (#3861)@lykGH , can you attach screenshots of the Device Manager with the device attached, as described in #3785 ? Also  - what FW version did use to upgrade the device ?
You can also run rs-enumerate-devices tool found in the Windows installer and attach the log.

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@IykGH Do you mean that you disconnect the camera during the firmware update? If so, please connect the camera and open the DFU tool to select "3: Recover devices stuck in Upgrade Mode." You should see your stuck camera there and recover it. Please make sure the camera connected during recover.

DFU can be downloaded from https://downloadcenter.intel.com/download/27514

lykGH, 2019-05-05:
I've tried it.it shows No devices are stuck in DFU mode.i have connected my device. -------- 原始邮件 --------发件人： RealSense Customer Support <notifications@github.com>日期： 今天 中午12:02收件人： IntelRealSense/librealsense <librealsense@noreply.github.com>抄送： lykGH <846296206@qq.com>, Mention <mention@noreply.github.com>主    题： Re: [IntelRealSense/librealsense] D415 is not recognized after firmware update (#3861)
@IykGH Do you mean that you disconnect the camera during the firmware update? If so, please connect the camera and open the DFU tool to select "3: Recover devices stuck in Upgrade Mode." You should see your stuck camera there and recover it. Please make sure the camera connected during recover.
DFU can be downloaded from https://downloadcenter.intel.com/download/27514

RealSenseCustomerSupport, 2019-05-05:
----------------------------------------------

@lykGH  Can you see RealSense USB devices in device manager? Please provide a snapshot if possible. Or if you can find a Ubuntu machine, please provide the output of "lsusb | grep 8086" in Ubuntu machine?

lykGH, 2019-05-05:
This is what my Device Manager displays.-------- 原始邮件 --------发件人： RealSense Customer Support <notifications@github.com>日期： 2019年5月5日周日 下午2:44收件人： IntelRealSense/librealsense <librealsense@noreply.github.com>抄送： lykGH <846296206@qq.com>, Mention <mention@noreply.github.com>主    题： Re: [IntelRealSense/librealsense] D415 is not recognized after firmware update (#3861)
@lykGH  Can you see RealSense USB devices in device manager? Please provide a snapshot if possible. Or if you can find a Ubuntu machine, please provide the output of "lsusb | grep 8086" in Ubuntu machine?

RealSenseCustomerSupport, 2019-05-06:
----------------------------------------------

@lykGH  Could you please attach again? I couldn't see the snapshot of your device manager.

closed 2019-05-12 by RealSenseCustomerSupport

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

@lykGH Could you please attach again? Thanks!

lykGH, 2019-05-14:
I put the picture in the attachment

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@lykGH Did you still have the issue? Could you please update if any progress? Thanks!

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@lykGH Didn't get response from your side for weeks. Will close the ticket if no other questions. Thanks!