realsenseai/librealsense #3862: Extracting 3-D world coordinae for all the pixels using D435

kyrs, 2019-04-28:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 5.09.02 | 
| Operating System & Version |   Win 10 | 
| Platform                                 | PC  |
| SDK Version                            |   2.17.1                           | 
| Language                            |  python                         | 
| Segment			|  Robot                          | 

### Issue Description
Hey,
I am trying to extract the 3d world coordinates for each of the pixel captured by the camera. 
Currently, I am doing so by looping over each of the pixel value and saving the corresponding world coordinate. 
```
                                row,col = depth_image.shape
				zthImage = np.zeros((row,col,3))
				for i in range(row):
					for j in range(col):
						depth = aligned_depth_frame.get_distance(i,j)
						print(depth)
						zthImage[i,j] = rs.rs2_deproject_pixel_to_point(
		                                depth_intrin, [i,j], depth)
						print(i,j)
				print(zthImage)		
				depthOfImage.append(zthImage)		
```
It works but drastically impact the frame processing rate of the  camera. 
Is there any way of getting 3-D coordinates for all the pixel values without affecting the frame rate.Something similar to  get_data function in ``` aligned_depth_frame.get_data() ``` ??

MartyG-RealSense, 2019-04-28:
The link below has a Python script for transforming pixels from a depth image to world coordinates.  I hope it will be helpful to you.

https://github.com/IntelRealSense/librealsense/issues/1904#issuecomment-398434434

MartyG-RealSense, 2019-04-28:
I wonder if Intel's Python tutorial on converting RGBD data to a 3D point cloud with XYZ coordinates may be helpful to you.

https://github.com/dorodnic/binder_test/blob/master/pointcloud.ipynb

RealSenseCustomerSupport, 2019-05-10:
----------------------------------------------

Hi kyrs,

As MartyG-RealSense suggested, not sure did it help? Any update?

Thanks!

RealSenseCustomerSupport, 2019-05-22:
----------------------------------------------

Hi kyrs,

Do you have any update on this one? Still need help?

Thanks!

kyrs, 2019-05-23:
@RealSenseCustomerSupport  I am still experimenting it out, will update it in couple of days.

Resays, 2019-06-10:
Here is an example to get RGB data from point how to get [XYZ RGB data from RealSense](https://github.com/Resays/xyz_rgb_realsense/tree/master)

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@kyrs  Any good news now? Looking forward to your update. Thanks!

closed 2019-06-17 by kyrs