realsenseai/librealsense #3864: python 3, mac os, realsense2, dylib, pyrealsense

ShenZehong, 2019-04-29:
Hi,
I've successfully install `librealsense2.2.20.0.dylib`, and run several examples in `mac 10.14.4`.

First: there're some errors when installing according to your document, especially in China(due to bad internet connection).
- when install `vulkan-sdk`  according to `https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md`, brew-cask cannot find this file. METHOD: install according to `https://vulkan.lunarg.com/doc/sdk/1.1.106.0/mac/release_notes.html`
- `make -j` is very slow and I met the error of `ld: library not found for -lusb-1.0`. The method in the document doesn't fit for `zsh`, you need manually add `export LIBRARY_PATH=/usr/local/lib:$LIBRARY_PATH` to your `~/.zshrc`

Second: HOWEVER, I REALLY WANT TO KNOW, where is the function list of the dylib? or how can I use it?(especially with `ctype` in `python3`?)
I found your product very tricky for mac with python3, while this is almost among the top 3 choices(or top 2) for programmer?? Is there a fully accomplished document going to be released? Thank you very much!

ShenZehong, 2019-04-29:
And is there any possibility of a distributed `pyrealsense2` in the future? I found some versions but not correct the platform~(just ask)

RealSenseCustomerSupport, 2019-05-07:
----------------------------------------------

@ShenZehong

1. For the error of `ld: library not found for -lusb-1.0,`there's  note in https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_osx.md as below
**Note :** On some Mac systems you might encounter `ld: library not found for -lusb-1.0` error (either in the terminal during make or in XCode) This can be worked-around by setting environment variable: `/bin/launchctl setenv LIBRARY_PATH /usr/local/lib`

2.  If you want to build Python bindings for the SDK, please run the top level CMake command with the following additional flag `-DBUILD_PYTHON_BINDINGS=bool:true`

For `pyrealsense2` support on MacOS, we will evaluate to see if possible. Thanks!

ShenZehong, 2019-05-08:
@RealSenseCustomerSupport 
Many thanks!
It works well on python3 on mac so far! 
Thanks again!! And you really need to update your documents~

closed 2019-05-08 by ShenZehong