realsenseai/librealsense #3865: There are two ways to get depth values. Which is more accurate?

chaojizhangdaxin, 2019-04-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 14/16/17)  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC|
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {opencv/python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
The first method is:
        self.depth_frame = self.aligned_frames.get_depth_frame()
        color_frame = self.aligned_frames.get_color_frame()
        # if not depth_frame or not color_frame:
        #     continue
        print('depth: ',self.depth_frame)
        print(self.depth_frame.get_distance(320,105))
The second method is:
depth_image = np.asanyarray(aligned_depth_frame.get_data())
depth = depth_image.astype(float)
depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()
depth = depth * depth_scale
depth = depth[320,105]
print(depth)

Which of the two methods gets the value closer to the actual distance?
[image]
Which method error is indicated in this picture? Thank you.

dorodnic, 2019-04-29:
Both methods should produce the exact same value. `get_distance` is more readable, but slower

chaojizhangdaxin, 2019-04-30:
> 两种方法都应该产生完全相同的值。`get_distance`更可读，但更慢

Thank you for your answer. I would like to ask why there is an error in my camera. The camera was mounted 3.5 meters away from the object. The camera I used was a D435. How to reduce the error? How to reduce the depth is 0 meters?

chaojizhangdaxin, 2019-04-30:
> 两种方法都应该产生完全相同的值。`get_distance`更可读，但更慢

谢谢你的回答，我想问一下为什么我获取深度存在误差。相机架设在距离物体3.5米的地方。我使用的相机是D435。怎样去减少误差？怎么样减少深度是0米的情况？

chaojizhangdaxin, 2019-04-30:
> 两种方法都应该产生完全相同的值。`get_distance`更可读，但更慢

Thank you for your answer. I would like to ask why there is an error in my depth acquisition. The camera was mounted 3.5 meters away from the object. The camera I used was a D435. How to reduce the error? How to reduce the depth is 0 meters?

dorodnic, 2019-04-30:
For getting best depth accuracy please take a look at ["Tuning depth cameras for best performance" white-paper](https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance)

chaojizhangdaxin, 2019-04-30:
> 要获得最佳深度精度，请查看[“调整深度相机以获得最佳性能”白皮书](https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance)

How do I find the white paper method in the python package

RealSenseCustomerSupport, 2019-05-07:
----------------------------------------------

@chaojizhangdaxin Could you please clarify more details about the error you mentioned? Snapshot will be helpful for us. The white paper is using RealSenseViewer for example. And you can also call the related API in python to achieve that.

chaojizhangdaxin, 2019-05-08:
> @chaojizhangdaxin您能否澄清一下您提到的错误的更多细节？快照对我们很有帮助。白皮书以RealSenseViewer为例。你也可以在python中调用相关的API来实现它。

The information in the image refers to which of two ways to get the depth value

chaojizhangdaxin, 2019-05-08:
19/5000  
I want my depth value to be closer to the actual distance!

RealSenseCustomerSupport, 2019-05-08:
----------------------------------------------

@chaojizhangdaxin Sorry I couldn't see the picture of your case. As you mentioned "I would like to ask why there is an error in my depth acquisition. The camera was mounted 3.5 meters away from the object. The camera I used was a D435. How to reduce the error? How to reduce the depth is 0 meters?"   Could you please provide the detailed snapshot you mentioned here?  Thanks!

抱歉，我们这边无法看到您实际测试结果的error(如果我理解正确的话，前面的图应该只是白皮书的截图，而不是您实际测试的结果图)。正如您之前提到的"I would like to ask why there is an error in my depth acquisition. The camera was mounted 3.5 meters away from the object. The camera I used was a D435. How to reduce the error? How to reduce the depth is 0 meters?" 。请提供您实际测试的结果图，以帮助我们理解您的问题并提供相应的支持。谢谢！

chaojizhangdaxin, 2019-05-08:
[image]
[image]

chaojizhangdaxin, 2019-05-08:
> @chaojizhangdaxin抱歉，我看不到你的案子的照片。如你所说“我想问为什么我的深度采集有误。相机安装在离物体3.5米处。我使用的相机是D435。如何减少误差？如何减小深度是0米？“ 你能提供你在这里提到的详细快照吗？谢谢！
> 
> 抱歉，我们这边无法看到您实际测试结果的错误（如果我理解正确的话，前面的图应该只是白皮书的截图，而不是您实际测试的结果图）。正如您之前提到的“我想问为什么我的深度采集有误。摄像机安装在离物体3.5米处。我使用的摄像机是D435。如何减少误差？如何减小深度为0米？。请提供您实际测试的结果图，以帮助我们理解您的问题并提供相应的支持谢谢！

This is a close-up.

chaojizhangdaxin, 2019-05-08:
[image]
This is a picture taken from a distance, but without any depth. The distance will cause errors between the depth value and the actual distance. You're going to get 0.

chaojizhangdaxin, 2019-05-08:
[image]

RealSenseCustomerSupport, 2019-05-08:
----------------------------------------------

@chaojizhangdaxin Could you please provide the depth image snapshot? From previous provided color image, we still couldn't see what's the issue there. Please capture a snapshot with RealSenseViewer which can be downloaded from https://github.com/IntelRealSense/librealsense/releases/download/v2.21.0/Intel.RealSense.Viewer.exe

chaojizhangdaxin, 2019-05-08:
[image]
[image]

chaojizhangdaxin, 2019-05-08:
> @chaojizhangdaxin你能提供深度图片快照吗？从以前提供的彩色图像，我们仍然无法看到那里的问题。请使用RealSenseViewer捕获快照，可以从https://github.com/IntelRealSense/librealsense/releases/download/v2.21.0/Intel.RealSense.Viewer.exe下载。[](https://github.com/IntelRealSense/librealsense/releases/download/v2.21.0/Intel.RealSense.Viewer.exe)

This is a picture taken in good light conditions

RealSenseCustomerSupport, 2019-05-08:
----------------------------------------------

@chaojizhangdaxin Sorry the depth image seems very dark to see clearly. Could you please capture a colorized depth image in realsenseviewer. You can print screen to save that.

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@chaojizhangdaxin Did you get through the issue now? Please update if you still need support. Thanks!

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@chaojizhangdaxin Sorry we didn't get feedback from you for weeks. Will close the ticket if no other questions.

closed 2019-06-03 by RealSenseCustomerSupport

chaojizhangdaxin, 2019-07-02:
Sorry, I've been looking at other projects. Didn't notice the mailbox. We now use other cameras. For the same scenario, the depth obtained by using SDK is more accurate. And the depth we get with our own programs is wrong.@RealSenseCustomerSupport