realsenseai/librealsense #3866: finding Z accuracy and plane fit RMS error python

scizors, 2019-04-29:
How can i find plane fit RMS error and Z accuracy using pyrealsense2 library in python. I have see the python wrappers but i need a help.

ev-mp, 2019-04-29:
@scizors , Plane fit RMS and Z accuracy are core elements of the Depth-Quality tool provided with the SDK. [link](https://github.com/IntelRealSense/librealsense/tree/master/tools/depth-quality)
The algorithmic part is embedded into the tool and not in the SDK (i.e. not included `pyrealsense2`).
The theoretic basis for that tool is established in the relevant [white paper](https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense_DepthQualityTesting.pdf)

In order to get that functionality in python you can reuse the core functionality from C++ by wrapping with python. You can also port and rewrite the algorithms into python.

scizors, 2019-04-29:
Thanks for the reply how about in C++ and please mention any other programming language to test the camera and its efficiency with respect to errors and various lighning condition I need to check it.

ev-mp, 2019-04-29:
You can use `rs-depth-quality` tool in a stand-alone manner in different environments (inside and outside) as long as the minimal requirements are met :
- A flat non-reflective surface is set in front of the camera
- You can hold camera stable enough to achieve the required perpendicularity threshold

Just to reiterate  - the methodology for testing the actual image quality is elaborated in the white paper link above. And you can follow that document (and/or actual C++ implementation) to introduce a similar functionality in python.

One thing to mention though  - using the `rs-depth-quality`  has the advantage that the tool has been internally validated, so you can use its results and generated reports to be acknowledged by the Realsense team.

scizors, 2019-04-29:
thanks for the reply

closed 2019-04-29 by scizors