realsenseai/librealsense #3867: GetOverlappedResult failed

Nkyoku, 2019-04-29:
| Required Info                         |         |
|--------------------------------|------ |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.0.18.5502 | 
| Operating System & Version | Win10 | 
| Platform                                 | PC |
| SDK Version                            | 2.20.0 | 
| Language                            |  Visual C++ 2017/2019  | 

### Issue Description
The following error messages are output 792 times at program start.
```
ERROR SensorDeviceIoControlWithTimeoutV2:GetOverlappedResult failed!:0x00000032
ERROR SensorGetDataFieldPropertiesV2:IOCTL_SENSOR_CLXV2_GET_DATA_FIELD_PROPERTIES:0x00000032
```

I encountered these error when I was trying 'pose' sample.
However, I found similar output when I attach debugger to RealSense Viewer.

RealSenseCustomerSupport, 2019-05-17:
----------------------------------------------

Hi @Nkyoku
Can you provide further details as to the system configuration you have setup?
- T265 plugged into USB3 port on host platform?  Or are you plugging T265 into a USB hub?
- You mention that RealSense Viewer and plugging in debugger, can you elaborate on this point a bit more?

Have you attempted the v2.21.0 instead of the v2.20.0?

It seems that this issue may have something to do with the winusb drivers.

Nkyoku, 2019-05-18:
Thank you for your response.

Firstly, I have connected T265 via USB3 into root hub of i5-8350U.

Secondly, I used Visual Studio to debug RealSense Viewer to see debug messages.
Because these messages appear during instantiating `rs2::context`, I had to attach debugger before the Viewer was initialized.

I have tried v2.21.0 at first, but I encountered error messages.
So I decided to downgrade SDK.
When I posted this issue, I was trying older v2.20.0. (but it seems same result)

After that, I have tested on other two computers.
One is Intel H170 chipset (Windows Server 2016) and another is AMD B450 (Windows 10).
There are no error messages on these two computers.
This problem may have been caused by my laptop.

closed 2019-06-17 by RealSenseCustomerSupport