realsenseai/librealsense #3868: How can i label 3D bounding box using RealSense D415 and C++

Resays, 2019-04-29:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
i'm actually looking to implement this research http://rgbd.cs.princeton.edu/
using RealSense D415 and c++
Successfully run the examples provided in SDK but not getting the idea how can i achieve the 3D bounding box using pointcloud example ... 
Anyone can help ?
@dorodnic @gayanbrahmanage

MartyG-RealSense, 2019-04-29:
My recollection is that drawing a 3D bounding box for a point cloud is more complex than a bounding box for a 2D image, as a three-dimensional line has to be drawn.  3D lines can be drawn on a point cloud using a **polyline**.  OpenCV can draw these with its **polylines()** function.

https://www.tutorialspoint.com/opencv/opencv_drawing_polylines.htm

The Python-based point cloud program Pyntcloud has an example where it demonstrates using polylines to draw axis lines on a point cloud.

https://github.com/daavoo/pyntcloud/blob/master/examples/%5Bvisualization%5D%20Polylines.ipynb

Resays, 2019-04-30:
Thanks for your reply, i can code using OpenCV and draw lines but the goal is to grab xyz position from rgb-d sensor and draw 3D environment as pointcloud is doing but the twist is, it should have bounding box around the objects.
But i am not getting the point, where should i start ?
the reference research is attached but not understanding how to do it ...
http://rgbd.cs.princeton.edu/

MartyG-RealSense, 2019-04-30:
I could not find a way to do 3D bounding boxes in OpenCV without using datasets and training data.  But since the example paper that you quoted uses these too, this may not be a problem for you.

I found an example project for generating such bounding boxes in OpenCV.

https://github.com/mittalrajat/3D-Bounding-Box-Estimation

Resays, 2019-05-02:
Thanks MartyG, yes i can use machine learning to detect objects 
and let me review the link first

closed 2019-05-03 by Resays