realsenseai/librealsense #3870: Callback Duration Error: Epoch Duration

AlexGraberTilton, 2019-04-29:
Frame Callback [Depth#0] overdue. (Duration: 1556562406681.580566ms, FPS: 30, Max Duration: 32ms)

https://github.com/IntelRealSense/librealsense/blob/bcc71f3b01c482dfa51895ab6247135c9c76323d/src/archive.cpp#L294

when frame is 0, get_frame_callback_start_time_point() is returning 0, which causes a overdue debug message due to duration being from epoch.

dorodnic, 2019-04-30:
Thanks @AlexGraberTilton 
Yes, this seems like a minor bug

Nir-Az, 2021-08-31:
Hi @AlexGraberTilton,

Thanks for notifying about this issue (and sorry for the late response).
We investigated the issue and we discovered that the prints were somehow confusing (You are right!).
The 0 you see that probably relates to a composite frame (a combo frame that holds several frames inside) which is missing metadata since each frame inside holds different data.

We are refactoring our logs for the frame lifetime currently.
I will update this ticket once we have a merged PR for it.

Nir-Az, 2021-09-05:
As I said, the logs has been refactored and should be much more user friendly now.
PR #9716 .
Will be contained in the next releaser.

I hope it helps.
Good luck.

RealSenseCustomerSupport, 2021-09-05:
----------------------------------------------

@AlexGraberTilton this issue has been addressed in the PR, https://github.com/IntelRealSense/librealsense/pull/9716 and we will close this ticket. If you have any new issues please feel free to open a new ticket.

closed 2021-09-05 by RealSenseCustomerSupport