realsenseai/librealsense #3871: realsense viewer failed to add deep camera D435i

IcarusXZ, 2019-04-30:
| Camera Model | D435i |
|**librealsense**|2.20.0 RELEASE|
|Operating System|Windows10|

Please provide a description of the problem
I followed the quick-start instructions  (https://www.intelrealsense.com/get-started-depth-camera/ )
but the real-sense viewer failed to add Depth camera D435i.By the way ,the depth camera was installed successfully.The error is illustrated below.
[image]

ev-mp, 2019-04-30:
@IcarusXZ , the error is raised by MS UVC driver and in general indicate some kernel/hw issue.  Note that it is often the case that the issue will disappear after re-plugging the camera and switching/checking the USB port and cable.

Also  - what did you mean by `the depth camera was installed successfully`  ? 

I would suggest to download the SDK (installer) and run `rs-enumerate-devices` to get get specific data

IcarusXZ, 2019-05-15:
@ev-mp ,Thank you ! I solved the problem by modifying Windows privacy options and allowing apps to access the camera.

closed 2019-05-15 by IcarusXZ