realsenseai/librealsense #3872: software device in Python

UserLiubov, 2019-04-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435                                                       | 
| Firmware Version                   | 05.10.03.00                                             | 
| Operating System & Version |   Win 10                                              | 
| Kernel Version (Linux Only)    |  -                                     | 
| Platform                                 | PC             |
| SDK Version                            |  SDK 2.0                          | 
| Language                            |  python                       | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I want to process raw depth data that I saved on my PC (I got them using D435). 
I found that there was software device class in C++ for this purpose. But I couldn't find anything similar in Python. Is there any functionality in Python to fill holes on raw depth data?
Thanks for your helping.

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

Hi UserLiubov ,

If for recorded bag file, you can refer to this example on how to deal with bag files:
https://github.com/IntelRealSense/librealsense/blob/79dda7f00ebde9564f1a2053b8ab95b7ba41ac69/wrappers/python/examples/read_bag_example.py

If for using filters in python code, you may refer to following example and code:
1. https://github.com/IntelRealSense/librealsense/blob/79dda7f00ebde9564f1a2053b8ab95b7ba41ac69/wrappers/python/examples/box_dimensioner_multicam/box_dimensioner_multicam_demo.py
2. Hole filtering
https://github.com/IntelRealSense/librealsense/blob/79dda7f00ebde9564f1a2053b8ab95b7ba41ac69/wrappers/python/python.cpp#L571

Thanks!

RealSenseCustomerSupport, 2019-05-23:
----------------------------------------------

Hi UserLiubov ,

Wonder if there is any update on this one? Or do you still need help?

Thanks!

closed 2019-06-06 by RealSenseCustomerSupport