realsenseai/librealsense #3876: uvcvideo mostly silent failure

j-gowdy, 2019-04-30:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | R435 | 
| Firmware Version                   | 05.11.06.200 | 
| Operating System & Version |   Ubuntu 16.04 | 
| Kernel Version (Linux Only)    |  4.14.111-139                                         | 
| Platform                                 | Odroid Xu-4  |

### Issue Description
I am trying to set up software for a Realsense D435 on a Odroid XU4 with  4.14.111-139, and I am seeing similar problems to the closed issue #1979.  
I followed the installation procedure, i.e.,
```
./scripts/patch-realsense-ubuntu-odroid-xu4-4.14.sh
sudo udevadm control --reload-rules && udevadm trigger
sudo modprobe uvcvideo
```
All seemed to succeed with no errors, but when I plug in the D435, I get
```
[ 5192.238758] usb 4-1.1: new SuperSpeed USB device number 6 using xhci-hcd
[ 5192.259550] usb 4-1.1: New USB device found, idVendor=8086, idProduct=0b07
[ 5192.259557] usb 4-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 5192.259562] usb 4-1.1: Product: Intel(R) RealSense(TM) Depth Camera 435 
[ 5192.259567] usb 4-1.1: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435 
[ 5192.259572] usb 4-1.1: SerialNumber: 812113022100
[ 5192.268625] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435  (8086:0b07)
[ 5192.269393] uvcvideo: Unable to create debugfs 4-6 directory.
[ 5192.269504] uvcvideo 4-1.1:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
[ 5192.269511] uvcvideo 4-1.1:1.0: Entity type for entity Processing 2 was not initialized!
[ 5192.269517] uvcvideo 4-1.1:1.0: Entity type for entity Camera 1 was not initialized!
[ 5192.269725] input: Intel(R) RealSense(TM) Depth Ca as /devices/platform/soc/soc:usb3-0/12000000.dwc3/xhci-hcd.3.auto/usb4/4-1/4-1.1/4-1.1:1.0/input/input8

```
Issue #1979 implies you get these messages when the proper uvcvideo patches are not applied, but,  I find that my newly built odroid_bionic/drivers/media/usb/uvc/uvcvideo.ko is the same file as /lib/modules/4.14.111-139/kernal/drivers/media/usb/uvc/uvcvideo.ko, and digging further I see that the patched files should have a new string "Depth data 16-bit", and when I run strings on the .ko files (in both places), that new string is there, so the patch has been applied and appears installed.
Any ideas as to what to do next?

ev-mp, 2019-04-30:
@j-gowdy , it is probably misinterpreted as the messages you posted are related just to the internal UVC node build-up. They are not indicative of the kernel patch being (or being not) applied.
The relevant lines in #1979 that are screaming - "patch required" are those:
>[ 4046.059731] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
[ 4046.059745] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71

But as mentioned - these are not related and can be safely disregarded

j-gowdy, 2019-04-30:
@ev-mp , thanks for the reassurance.   I don't get those same alarming "Entity type" warnings when I use the DKMS drivers under Ubuntu 16.04 on a normal Intel processor with the D435, so I was worried.   I am getting depth and color when I use rs-capture, and in my own custom application, but the device stops giving me images after 30 minutes or so on the ARM (where the same application runs indefinitely on the normal Intel processor), so these warnings were high on my list of "things that are different" that might be the underlying cause.   I have more poking to do before I can ask a useful question about my real problem.  Thanks again.

ev-mp, 2019-04-30:
@j-gowdy , thanks for confirming.
I will close it if no follow ups on that log is needed.

A note regarding the 30 min threshold - monitoring the Odroid's CPU temperature/ throttling and checking `dmesg` logs may provide further hints.

closed 2019-05-01 by ev-mp