realsenseai/librealsense #3877: Compressing depth frames exported to image files: can colorizer be used?

realsens01, 2019-04-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.1 | 
| Operating System & Version |   Windows 10 / Ubuntu 18.10 | 
| Kernel Version (Linux Only)    |  NA / 4.18.0-17-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.20.0                        | 
| Language                            | Python                          | 
| Segment			|  NA                          | 

### Issue Description
Realsense bag files are very large. We are collecting a lot of data. I'd like to use lossy compression to store the frames extracted from a bag file, so that I can delete the original bags. For the color stream, JPEG conversion substantially reduces the required storage, but this cannot be applied to the 16 bit format of the depth files without shrinking the dynamic range by 2 orders of magnitude. TIFF compression does not seem to shrink the file size substantially, but the conversion tools I'm using (e.g., Imagemagick) do not seem to support JPEG 2000. Dropping to 8 bits seems like it would make for a much more substantial loss of information, relative to the loss incurred by standard compression techniques. Am I wrong about this? If not, then are there any other options for storing 16 bit grayscale using lossy compression?

I'm wondering if the librealsense colorizer could be useful. Is it possible to convert the grayscale images, using the colorizer, when the frames have already been exported to image files (i.e., they are not wrapped in a frame object)? I'm wondering if this would make it possible to apply standard JPEG compression to 8x3 bit colorized images, without sacrificing dynamic range.

I don't know a great deal about image and video storage. Any tips will be appreciated.

realsens01, 2019-05-01:
This might not be an issue anymore. It requires further testing, but I've gotten some improved results by adjusting TIFF compression settings.

RealSenseCustomerSupport, 2019-05-10:
----------------------------------------------

Hi @realsense01,

You can also try to compress rosbag file. There is one option in the setting of realsense-viewer.

closed 2019-06-03 by dorodnic