realsenseai/librealsense #3878: Recording both infrared streams via the Python wrapper

realsens01, 2019-04-30:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 5.11.1 | 
| Operating System & Version |   Windows 10 / Ubuntu 18.10 | 
| Kernel Version (Linux Only)    |  NA / 4.18.0-17-generic                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.20.0                        | 
| Language                            | Python                          | 
| Segment			|  NA                          | 

### Issue Description
Using the Python API, I would like to record a bag file that contains both IR streams. I thought this would be straightforward, based on prior experience, but something is going wrong. My code is very simple: it uses a config object to enable the streams that I'm interested in recording from, and then passes the config to pipeline.start. It works, in general, but fails when I try to add a second IR stream (telling me that there is a problem enabling a stream). What can I do?

Some notes:
* The API How-To suggests [using enable_stream with stream indices 0 and 1](https://github.com/IntelRealSense/librealsense/wiki/API-How-To#start-streaming-left-and-right-infrared-imagers). I tried that (using Python, not C++), and it returned the same error.
* A reply to another user question suggests [using enable_stream with stream indices 1 and 2](https://github.com/IntelRealSense/librealsense/issues/1140#issuecomment-364737911). This also didn't work. The only stream index that works is 1.
* Recording the IR streams with Realsense Viewer worked.
* ``rs2.context().devices[0].sensors[0].profiles`` contains profiles for ``Infrared(1)`` and depth only. There are no other infrared streams. I'm not sure if this is relevant.

ev-mp, 2019-04-30:
@realsens01 , this commonly occurs when the device is enumerated as USB2. As with USB2 the device is designed to provide only a subset of streaming modes.
Run `realsense-viewer` or `rs-enumerate-devices -s` to check the actual connection type details.

The other less common reason is that device's "advanced mode" is being turned off. Both tools will display the "advanced mode" status, with the Viewer being also able to toggle it if necessary.

realsens01, 2019-05-02:
Thanks @ev-mp . I used Realsense Viewer to confirm that the device is recognized as USB 3.0 and that advanced mode is enabled. The problem persists.

Here is some sample output in an IPython terminal:
```
Python 3.6.5 (v3.6.5:f59c0932b4, Mar 28 2018, 17:00:18) [MSC v.1900 64 bit (AMD64)]
Type 'copyright', 'credits' or 'license' for more information
IPython 6.3.1 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import pyrealsense2 as rs2

In [2]: device = rs2.context().devices[0]
   ...: serial_number = device.get_info(rs2.camera_info.serial_number)
   ...: config = rs2.config()
   ...: config.enable_device(serial_number)
   ...: config.enable_stream(rs2.stream.infrared, 0)
   ...: config.enable_stream(rs2.stream.infrared, 1)
   ...: pipeline = rs2.pipeline()
   ...: pipeline.start(config)

---------------------------------------------------------------------------
RuntimeError                              Traceback (most recent call last)
<ipython-input-2-01583c32427f> in <module>()
      9 pipeline = rs2.pipeline()
---> 10 pipeline.start(config)

RuntimeError: Config couldn't configure all streams

In [4]: device.get_info(rs2.camera_info.advanced_mode)
Out[4]: 'YES'
```
I got the same result using indices 1 and 2, instead of 0 and 1. Fwiw, here are the available stream profiles:
```
In [5]: device.sensors[0].profiles
Out[5]:
[<pyrealsense2.video_stream_profile: Infrared(1) 1280x720 @ 30fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 1280x720 @ 15fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 1280x720 @ 6fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 848x480 @ 90fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 848x480 @ 60fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 848x480 @ 30fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 848x480 @ 15fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 848x480 @ 6fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 640x480 @ 90fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 640x480 @ 60fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 640x480 @ 30fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 640x480 @ 15fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 640x480 @ 6fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 640x360 @ 90fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 640x360 @ 60fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 640x360 @ 30fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 640x360 @ 15fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 640x360 @ 6fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 480x270 @ 90fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 480x270 @ 60fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 480x270 @ 30fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 480x270 @ 15fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 480x270 @ 6fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 424x240 @ 90fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 424x240 @ 60fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 424x240 @ 30fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 424x240 @ 15fps Y8>,
 <pyrealsense2.video_stream_profile: Infrared(1) 424x240 @ 6fps Y8>,
 <pyrealsense2.video_stream_profile: Depth(0) 1280x720 @ 30fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 1280x720 @ 15fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 1280x720 @ 6fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 848x480 @ 90fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 848x480 @ 60fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 848x480 @ 30fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 848x480 @ 15fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 848x480 @ 6fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 640x480 @ 90fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 640x480 @ 60fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 640x480 @ 30fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 640x480 @ 15fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 640x480 @ 6fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 640x360 @ 90fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 640x360 @ 60fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 640x360 @ 30fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 640x360 @ 15fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 640x360 @ 6fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 480x270 @ 90fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 480x270 @ 60fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 480x270 @ 30fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 480x270 @ 15fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 480x270 @ 6fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 424x240 @ 90fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 424x240 @ 60fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 424x240 @ 30fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 424x240 @ 15fps Z16>,
 <pyrealsense2.video_stream_profile: Depth(0) 424x240 @ 6fps Z16>]
```

realsens01, 2019-05-02:
This issue might be the same as https://github.com/IntelRealSense/librealsense/issues/3885

realsens01, 2019-06-06:
Any suggestions @RealSenseCustomerSupport? I'd really like to be recording this data.

RealSenseCustomerSupport, 2019-06-19:
----------------------------------------------

From your video profile output it does looks like the Advanced mode is not enabled.
Is the IR 2 available via realsense-viewer?

realsens01, 2019-06-19:
@RealSenseCustomerSupport

* I confirmed that Advanced Mode is enabled.
* Both IR streams are available in Realsense Viewer

Both of these things were already stated in this short thread.

closed 2019-07-03 by RealSenseCustomerSupport

realsens01, 2019-07-03:
@RealSenseCustomerSupport : Why would you close this? This is the 3rd issue that I've had that you've closed without resolving. I've been working with RealSense for about a year or more now on a funded project and I've literally never gotten a solution from you.

bwilfley, 2019-12-29:
I stumbled across the following that seems to work. The upshot is that the IR cameras are numbered 1 and 2, not 0 and 1 (in the calls to enable_streams).

```
#%% imports
import pyrealsense2 as rs

#%% configure
ctx = rs.context()
device = ctx.devices[0]
serial_number = device.get_info(rs.camera_info.serial_number)
config = rs.config()
config.enable_device(serial_number)

#%% enable streams
config.enable_stream(rs.stream.infrared, 1, 1280,720, rs.format.y8, 6)
config.enable_stream(rs.stream.infrared, 2, 1280,720, rs.format.y8, 6)

#%% run pipeline
pipeline = rs.pipeline()
profile = pipeline.start(config)
```

vstadnytskyi-FDA, 2023-04-04:
> I stumbled across the following that seems to work. The upshot is that the IR cameras are numbered 1 and 2, not 0 and 1 (in the calls to enable_streams).
>
> ```
> #%% imports
> import pyrealsense2 as rs [...]

Here is an explicit way to configure color, depth and IR streams specifically for D455
```
conf.enable_stream(stream_type=rs.stream.color, width = 1280, height = 720, format = rs.format.rgb8, framerate = 30)
conf.enable_stream(stream_type=rs.stream.depth, width = 848, height = 480, format = rs.format.z16,framerate = 30)
conf.enable_stream(stream_type=rs.stream.infrared, stream_index = 1, width = 848, height = 480, format = rs.format.y8,framerate = 30)
conf.enable_stream(stream_type=rs.stream.infrared,stream_index = 2, width = 848, height = 480, format = rs.format.y8,framerate = 30)

```