realsenseai/librealsense #3879: Win10 D435 Error

Arce0406, 2019-05-01:
| | |
|---|---|
|**librealsense**|2.21.0 RELEASE|
|**OS**|Windows|

I got a error message when open Intel RealSense Viewer, 
----------------------------------------------------------------------------------------------
Backend in rs2_create_device(info_list:000002966E9E8860, index:0):
MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x80070005: "X֫�"
----------------------------------------------------------------------------------------------

I tried re-install the sdk (version 2.21.0), doesn't worked.
And win10 version is 1803, and I do have install Kinect SDK 1.8 (I think this is not problem), no CUDA、OPEN CV.
Could anyone has same problem? or solutions about this error?

ev-mp, 2019-05-01:
@Arce0406 hi , this MS error code stands for "Access denied".
This error was reported in the past (#3644,#3381) and handled by updating core MS components.

GregBahm, 2019-05-02:
I am getting the same error in unity. Also with a D435 on Windows 10.
When I open the Realsense Viewer 2.20.0, It tells "Firmware update recommended. Current firmware version 05.09.02.0. Minimal firmeware version 5.10.03.00." When I download the recommended update from [here](https://downloadcenter.intel.com/download/28724/Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t) and try to install it with [this firmware update tool](https://downloadcenter.intel.com/download/27514/Windows-Device-Firmware-Update-Tool-and-Latest-Firmware-for-Intel-RealSense-D400-Product-Family?v=t), I get the message "No DFU device was connected
Failed to burn firmware"

The 415 works fine in Unity on my computer.

Arce0406, 2019-05-03:
Thanks for you guys, after read your links, I found problem is the privacy of camera and location, after open this two setting everything works fined! gosh...

closed 2019-05-13 by ev-mp