realsenseai/librealsense #3881: OpenNI2_tracker with d435 in ROS

kongwu1993, 2019-05-01:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I don't know if this is the right place to ask. But I wonder if there is a way to run [openni2_tracker](https://github.com/ros-drivers/openni2_tracker) in ROS with d435. I've followed the openni2 wrapper [here](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/openni2) and the SimpleViewer works. I then installed NiTE and the skeleton tracking also works by running ./UserViewer. Finally I installed [openni2_tracker](https://github.com/msr-peng/openni2_tracker), however running the launch file always ends up "device not found".

Now my hunch tells me there must be a way of accessing the camera via the librealsense OpenNI2 wrapper and then publishing the skeleton coordinates in ROS. Has anyone attempted this before? Any pointers would be appreciated!

MartyG-RealSense, 2019-05-02:
The discussion in the link below offers methods to try if you get 'device not found' with OpenNI2 with a D415 or D435 camera.

https://github.com/IntelRealSense/librealsense/issues/2697

kongwu1993, 2019-05-02:
Hi Marty, thx for the link. I already followed that and the wrapper is working with d435. I just wonder how to integrate that into ROS. For librealsense openni2 wrapper we need to  move some .so files around, how would that correspond to openni2_tracker when it so far only works with Xtion cameras?

MartyG-RealSense, 2019-05-02:
As a general principle, if you are using a development environment (Visual Studio, Unity, Android Studio, etc) and you need to move files around then it is best to move them within the development environment instead of drag and drop outside of the environment.  If the files are moved within the tool then it can automatically update its linkages, but if the moving is done outside of the development environment then the linkages to files will be out of date and the environment will no longer know where you find them unless you define links to their new locations manually.

RealSenseCustomerSupport, 2019-05-09:
----------------------------------------------

@kongwu1993 Did you get it through? Looking forward to your update. Thanks!

kongwu1993, 2019-05-09:
Not really. I'm looking for specifics of integrating it with OpenNi in ROS, a firmware rather than development environment. Guess I'm at the wrong place to ask, will go to ROS openNi2 repo. Thanks anyway!

closed 2019-05-09 by kongwu1993

AlessandroMelino, 2021-03-05:
Hello @zinan93.

I would like to reopen this issue to ask you if it finnaly worked for you.

I am facing a similar issue trying to use the openni2.launch of ROS with a Realsense D435. So when I execute this, appears in terminal:

`[ INFO] [1614940981.625499350]: No matching device found.... waiting for devices. Reason: std::string openni2_wrapper::OpenNI2Driver::resolveDeviceURI(const string&) @ /tmp/binarydeb/ros-noetic-openni2-camera-1.5.1/src/openni2_driver.cpp @ 742 : Invalid device number 1, there are 0 devices connected.`

I hope you can help me. Thanks in advance.

Best regards.
Alessandro