realsenseai/librealsense #3885: Unable to get the left and the right infrared streams from D435 with Paython

PhilippRay, 2019-05-02:
The color and depth is working fine.
Even when i only want 1 input of infrared.

But when the output / video is running -> only the right IR channel is used.

how do i get both?
And the depth map is working over both left and right ...

PhilippRay, 2019-05-02:
i tried this one .. but that doesnt work!
config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 6)
config.enable_stream(rs.stream.infrared, 2, 640, 480, rs.format.y8, 6)

The Error says: Runtime Error: Couldnt resolve requests

dorodnic, 2019-05-02:
Perhaps the camera is not in advanced mode? 
Please open the Viewer and select one of the presets. If a popup will appear, say yes, the camera will restart and enable Infrared2 formats.

PhilippPoeschke, 2019-05-02:
is there a opportunity to change that in the code? i mean activate the advanced mode and activate the second infrared stream?

realsens01, 2019-05-02:
This seems related to my issue https://github.com/IntelRealSense/librealsense/issues/3878

PhilippRay, 2019-05-03:
i dont know why it is working now but this one is working fine for me: 

```
pipeline = rs.pipeline()

config = rs.config()
config.enable_stream(rs.stream.infrared, 1, 640, 480, rs.format.y8, 30)
config.enable_stream(rs.stream.infrared, 2, 640, 480, rs.format.y8, 30)
config.enable_stream(rs.stream.color, 640, 480, rs.format.bgr8, 30)
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

colorizer = rs.colorizer()
pipeline.start(config)

profile = pipeline.get_active_profile()
infrared_profile = rs.video_stream_profile(profile.get_stream(rs.stream.infrared, 2))
infrared_intrinsics = infrared_profile.get_intrinsics()
depth_profile = rs.video_stream_profile(profile.get_stream(rs.stream.depth))
depth_intrinsics = depth_profile.get_intrinsics()

print(infrared_profile)
print(infrared_intrinsics)
print(depth_profile)
print(depth_intrinsics)

while True:

    frames = pipeline.wait_for_frames()

    infrared_frame_zero = frames.get_infrared_frame(1)
    infrared_frame_one  = frames.get_infrared_frame(2)
    depth_frame = frames.get_depth_frame()
    color_frame = frames.get_color_frame()

    infrared_colormap_zero = np.asanyarray(colorizer.colorize(infrared_frame_zero).get_data())
    infrared_colormap_one = np.asanyarray(colorizer.colorize(infrared_frame_one).get_data())
    depth_colormap = np.asanyarray(colorizer.colorize(depth_frame).get_data())
    colormap = np.asanyarray(colorizer.colorize(color_frame).get_data())

    images = np.hstack((depth_colormap, colormap))

    cv2.imshow('RealSense', images)

    if cv2.waitKey(25) == ord('q'):
        break
pipeline.stop()
cv2.destroyAllWindows()
```

RealSenseCustomerSupport, 2019-05-08:
----------------------------------------------

Hi PhilippRay,

Thanks for the update. Close this one.

realsens01, 2019-05-14:
@RealSenseCustomerSupport: PhilippRay's code does not work for me. Same problem. 

@dorodnic : As I mention in https://github.com/IntelRealSense/librealsense/issues/3878, advanced mode is enabled in my case, and both IR streams can be recorded via Realsense Viewer, but not via the Python interface.

closed 2019-05-15 by RealSenseCustomerSupport