realsenseai/librealsense #3886: Install camera driver automatically

EmmanuelCos, 2019-05-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 } | 
| Operating System & Version |  Win 10  | 

Hello,

I see that the driver to make the camera work on the computer can be installed by right clicking the file "RS400Camera.inf", then click on install. 
https://downloadcenter.intel.com/download/28723/Intel-RealSense-D400-Series-Universal-Windows-Platform-UWP-Driver-for-Windows-10?product=87466

Is there a way to install the driver automatically from a program outside?

Thank you and regards,

Emmanuel

MartyG-RealSense, 2019-05-02:
The UWP driver is designed to make the camera more compatible with applications that use Microsoft's Universal windows Platform (UWP).  An example of such an application is the Microsoft HoloLens mixed-reality headset.

The UWP driver is not required for the 400 Series cameras to function.  If you still need to install an INF file easily though, one way to do so is to create a **batch file**, a file that executes a list of instructions when run.  The link below has advice for setting the contents of a batch file to install an inf file.

https://stackoverflow.com/questions/14096347/how-to-install-inf-file-from-base-directory-from-batch-file-bat

Google 'run inf with batch file' for further suggestions about how to create an inf-installing batch file.

RealSenseCustomerSupport, 2019-05-14:
----------------------------------------------

Hi @EmmanuelCos,

Is the question clarified?

EmmanuelCos, 2019-05-14:
Yes it is, thank you!

closed 2019-05-23 by RealSenseCustomerSupport