realsenseai/librealsense #3887: error in export_to_ply() in pyglet_pointcloud_viewer.py

Leo16112, 2019-05-02:
Can not save ply in example. The rest works.  I hadn't change any piece of code in example.

Traceback (most recent call last):
  File "/pyglet_pointcloud_viewer.py", line 467, in <module>
    pyglet.app.run()
  File "/usr/local/lib/python3.6/dist-packages/pyglet/app/__init__.py", line 138, in run
    event_loop.run()
  File "/usr/local/lib/python3.6/dist-packages/pyglet/app/base.py", line 142, in run
    self._run()
  File "/usr/local/lib/python3.6/dist-packages/pyglet/app/base.py", line 154, in _run
    timeout = self.idle()
  File "/usr/local/lib/python3.6/dist-packages/pyglet/app/base.py", line 275, in idle
    redraw_all = self.clock.call_scheduled_functions(dt)
  File "/usr/local/lib/python3.6/dist-packages/pyglet/clock.py", line 309, in call_scheduled_functions
    item.func(dt, *item.args, **item.kwargs)
  File "/pyglet_pointcloud_viewer.py", line 461, in run
    points.export_to_ply('./out.ply', mapped_frame)
TypeError: export_to_ply(): incompatible function arguments. The following argument types are supported:
    1. (self: pyrealsense2.pyrealsense2.points, arg0: str, arg1: pyrealsense2.pyrealsense2.video_frame) -> None

Invoked with: <pyrealsense2.pyrealsense2.points object at 0x7fa5240b8a08>, './out.ply', <pyrealsense2.pyrealsense2.frame object at 0x7fa5240b8c70>

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D415 } | 
| Operating System & Version |   Ubuntu 18.04  | 
| Version of SDK                      |  pyrealsense2-2.21.0.776

RealSenseCustomerSupport, 2019-05-08:
----------------------------------------------

@Leo16112  The error was caused incompatible type. Please modify https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/pyglet_pointcloud_viewer.py#L391  to below and try again. Thanks!

other_frame = frames.first(other_stream)**.as_video_frame()**

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

@Leo16112 Did you get it through? Looking forward to your update. Thanks!

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@Leo16112 Will close this ticket if no other questions. Thanks!

Leo16112, 2019-05-21:
Hello, thank you for reply. It worked for me.

closed 2019-05-28 by RealSenseCustomerSupport