realsenseai/librealsense #3888: Error getting temperature with frame_metadata_value

TOMAxxHAWK, 2019-05-02:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   |5.11.1.100 | 
| Operating System & Version |   {Win (10)   |                                
| Platform                                 | PC|
| SDK Version                            | v2.19.0                         | 
| Language                            |  Matlab 2018a                        | 
| Segment			|  Logging                      | 

I need to log the temperature of the camera in order to see if it effects the accuracy of the camera. I am doing this in Matlab due to my limited knowledge of other languages. 

My currenct lines of code to sample it is:

{everything before this is the same as in the depth_example.mat}
```

for i =1:n
f = pipe.wait_for_frames();
T(n) = f.get_frame_metadata(realsense.frame_metadata_value(8)); %%8 is the temperature
...
end 
```

When I run this code I get the error message:

> 
> Error using librealsense_mex
> Temperature attribute is not applicable for Depth stream
> 
> Error in realsense.frame/get_frame_metadata (line 34)
>             metadata = realsense.librealsense_mex('rs2::frame',
>             'get_frame_metadata', this.objectHandle,
>             int64(frame_metadata));
> 

The currect variable f is now a 'frameset'. But the function seems to need a 'frame'. So what can I do to fix this?

ev-mp, 2019-05-02:
@tomaviram , the device temperature attribute metadata is defined for T265 and not available for D415.
As a general rule, metadata attributes are designed to be [probed and queried dynamically](https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md#librealsense2-api-for-metadata) to avoid unnecessary exceptions

D415 has two temperature sensors installed: one for the PCB and the other for the Laser projector:
 - RS2_OPTION_ASIC_TEMPERATURE
 - RS2_OPTION_PROJECTOR_TEMPERATURE

These sensors are available via the `sensor' object interface:
(Warning: C++)
```c++
        auto list = ctx.query_devices();
        auto dev = list[0];
        auto sensors = dev.query_sensors();
        for (auto s : sensors)
            if (s.supports(RS2_OPTION_ASIC_TEMPERATURE))
                auto tmp = s.get_option(RS2_OPTION_ASIC_TEMPERATURE);
       ...
```

Two things regarding the actual acquisition:
- The temperature sensors are activated only during streaming. 
- Avoid raising exceptions  (like in the case above)  by applying the same retrieval mode - probe->read:
```c++
if s.supports(RS2_OPTION_ASIC_TEMPERATURE)
 val = s.get_option(RS2_OPTION_ASIC_TEMPERATURE);
```

TOMAxxHAWK, 2019-05-02:
Thanks ev-mp, that explains a lot!

I will try and see if I can perform the above mentioned code in Matlab. Will the post the code here if it works in case somebody would like to log it in Matlab as well.

RealSenseCustomerSupport, 2019-05-09:
----------------------------------------------

@TOMAxxHAWK Did you get it through? Looking forward to your update. Thanks!

TOMAxxHAWK, 2019-05-09:
Apparently, the solution given by ev-mp didn't work in Matlab. My problem was, that the first camera I tried to log the ASIC temperature did't work. My first guess was to use the get_option(23). That camera has an older firmware version than my current one (5.11.1.100). The solution was actually quite easy:
`frame_rate = 15;
height = 720;
width = 1280;
pipe = realsense.pipeline();
settings = realsense.config();
settings.enable_stream(realsense.stream.depth,width,height,realsense.format(0),frame_rate); %Setting the framerate to the desired value
profile = pipe.start(settings);
device = profile.get_device();
sensor = device.first('depth_sensor');
Temp = sensor.get_option(23);
`

For some reason this didn't work in the beginning and that's why I started looking in the direction of Frame Metadata.

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

@TOMAxxHAWK Could you please provide the error details such as the error logs or snapshot?  Also please refer to ev-mp's comment to add some check to avoid exception. Thanks!

    if s.supports(RS2_OPTION_ASIC_TEMPERATURE)
     val = s.get_option(RS2_OPTION_ASIC_TEMPERATURE);

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@TOMAxxHAWK Could you please update? Thanks!

TOMAxxHAWK, 2019-05-21:
@RealSenseCustomerSupport @ev-mp 
Excuses for the late reply. Unfortanetly I have finished my research with the camera and am now focussing on writing the report. I hope I can provide some information on the error in a later stage. From what I remember is that the sensor function in Matlab, gave cell-structures and those are very difficult to use in Matlab. Because I got a device cell structure with 2 sensors in it. So I didn't know which one to use. After using the get_option on either one I was still getting errors. 

But after trying the function on my main camera I didn't get the error when using Get_option(). So I don't know what I did wrong in the beginning but the solution is quit easy actually as seen above.

RealSenseCustomerSupport, 2019-06-06:
----------------------------------------------

@TOMAxxHAWK So the issue is resolved, right?

closed 2019-06-13 by TOMAxxHAWK