realsenseai/librealsense #3890: Duplicate sequential frame numbers, and timestamps == 0

evercoast-dev, 2019-05-02:
Camera Model: D400
Firmware Version: 5.10.03, 5.11.6
Operating System: Ubuntu
Kernel Version: 4.15.0-43-generic (we see the problem across a variety of kernel versions)
Platform: PC
SDK Version: 2.19.0, 2.20.0
Language: C++
Segment: VR/AR

On Ubuntu with a multi-camera rig, usually 12 for testing, at 15hz and 30hz, we're seeing two things happening frequently that we're trying to understand:

1) Occasionally we're seeing duplicate device frame #'s following calling wait_for_frames(), e.g.:
        rs2::frameset frameset = devicePipeline.wait_for_frames(100);
        rs2::frame depth_frame = frameset.get_depth_frame();
        unsigned long long  depth_frame_number = depth_frame.get_frame_number()

        where depth_frame_number at t(n - 1) == depth_frame_number at t(n)        

        Best practice question: How does the wait duration correlate to the desired frame rate?

        Best practice question: What is the best queue configuration to minimize frame drops?
                i.e., we're presently setting: RS2_OPTION_FRAMES_QUEUE_SIZE to 0 (zero)

2) Occasionally we're seeing considerably out of order frames coming in with time stamps of 0 (zero). Here's a real world example, format frame #, timestamp:
                00814, 0000000078390.656
                00815, 0000000078424.256
                **00254, 00000000000000000**
                00817, 0000000078491.456
                00818, 0000000078525.056
                00819, 0000000078558.656
                **00255, 00000000000000000**
                00821, 0000000078625.856
                00822, 0000000078659.456
                00823, 0000000078693.056
                **00256, 00000000000000000        <- Here note that we have a duplicate out 
                00256, 00000000000000000**                of sequence frame
                00825, 0000000078760.256

Note 1: We're not seeing this issue on Windows.

Note 2: We're very confident in our power and USB bandwidth for our multi-camera systems.

Thank you for any help & advice!

evercoast-dev, 2019-05-03:
Additional information, we ran some tests:

Specific Ubuntu kernel version: 4.15.0-48-generic
We tried firmware versions 5.10.3, and 5.11.6, and both are fine. This bug seems related to either the uvcvideo driver or the library.

So, we went back to 2.16.0, and it works fine. 2.16.0 was a long-term development version for us on Ubuntu where we knew we weren't seeing this problem, since then we've been working with 2.19.0 and 2.20.0 on Windows -- we also note that this bug does not effect Windows.

We write out a per camera log file, as you can see, data is good:

F = our frame #
DF = device depth frame #
DSYS = device depth frame time-stamp
CF = device color frame #
CSYS = device color frame time-stamp
ISDT = inter-sensor delta time

`GOOD  |  F#: (00010)  |  DF#: (07600)  |  DSYS (01556897748873.87)  |  CF#: (07599)  |  CSYS (01556897748890.03)  |  ISDT (0016.165771484375)`
`GOOD  |  F#: (00011)  |  DF#: (07601)  |  DSYS (01556897748907.43)  |  CF#: (07600)  |  CSYS (01556897748923.64)  |  ISDT (0016.206298828125)`
`GOOD  |  F#: (00012)  |  DF#: (07602)  |  DSYS (00001556897748941)  |  CF#: (07601)  |  CSYS (01556897748957.19)  |  ISDT (0016.191162109375)`
`GOOD  |  F#: (00013)  |  DF#: (07603)  |  DSYS (01556897748974.68)  |  CF#: (07602)  |  CSYS (01556897748990.81)  |  ISDT (0016.122314453125)`
`GOOD  |  F#: (00014)  |  DF#: (07604)  |  DSYS (01556897749008.19)  |  CF#: (07603)  |  CSYS (01556897749024.41)  |  ISDT (0016.220947265625)`
`GOOD  |  F#: (00015)  |  DF#: (07605)  |  DSYS (01556897749041.79)  |  CF#: (07604)  |  CSYS (00001556897749058)  |  ISDT (0016.217041015625)`
`GOOD  |  F#: (00016)  |  DF#: (07606)  |  DSYS (001556897749075.4)  |  CF#: (07605)  |  CSYS (01556897749091.59)  |  ISDT (000016.1923828125)`
`GOOD  |  F#: (00017)  |  DF#: (07607)  |  DSYS (00001556897749109)  |  CF#: (07606)  |  CSYS (01556897749125.21)  |  ISDT (0016.212646484375)`
`GOOD  |  F#: (00018)  |  DF#: (07608)  |  DSYS (001556897749142.6)  |  CF#: (07607)  |  CSYS (01556897749158.82)  |  ISDT (0016.219970703125)`
`GOOD  |  F#: (00019)  |  DF#: (07609)  |  DSYS (01556897749176.19)  |  CF#: (07608)  |  CSYS (01556897749192.39)  |  ISDT (000016.1923828125)`
`GOOD  |  F#: (00020)  |  DF#: (07610)  |  DSYS (01556897749209.82)  |  CF#: (07609)  |  CSYS (01556897749225.99)  |  ISDT (000016.1630859375)`

Any help most appreciated!

evercoast-smm, 2019-05-29:
Wanted to refresh this as it's proving to be an incredibly persistent problem for us.

@dorodnic do you have any insight here? Is anyone else reporting out of order frames #'s coming in?

Again, we're looping each pipeline in _its own thread_. Where a thread is assigned a device, and the thread is run. Each thread then boots by creating its own pipeline based by configuring the device, and then each thread spins calling:

        frameset = devicePipeline.wait_for_frames(500);

Are there any thread safety concerns that we should be aware of?

Thanks for your attention, this is a serious problem for us. We're seeing this everywhere from small configurations, e.g. 2-3 cameras on a laptop, to desktop rigs with 6 or 8 cameras, up to workstation rigs with 20+ cameras.

evercoast-smm, 2019-05-30:
After spending a lot of engineering time, we can confirm that this issue seems to go away with kernel versions 4.16 and 4.18. We're actively testing and working to confirm.

This seems like a very serious bug...is there anybody out there?

evercoast-smm, 2019-05-30:
I want to also mention that we can verify that this problem only manifests itself in multi-camera, multi-thread scenarios.

RealSenseCustomerSupport, 2019-07-01:
----------------------------------------------

@evercoast-smm Did you mean that the issue not seen with kernel 4.16 and 4.18? Do you also get such issue if you use less cameras?

RealSenseCustomerSupport, 2019-07-09:
----------------------------------------------

@evercoast-smm If no other questions about this issue, will close it accordingly. Thanks!

closed 2019-07-09 by RealSenseCustomerSupport