realsenseai/librealsense #3891: What is the appropriate scope for rs2::context?

evercoast-dev, 2019-05-02:
Camera Model: D400
Firmware Version: 5.10.03+
Operating System: Ubuntu
Kernel Version: 4.15.0-43-generic (we see the problem across a variety of kernel versions)
Platform: PC
SDK Version: 2.19.0+
Language: C++
Segment: VR/AR

Best Practice Q: What is the appropriate scope of an rs2::context? Is it best to construct a context only once per application execution, or can a context be constructed on the stack at any point it's needed?

Most of the supplied examples are quite small applications. I ask in the context of a more complex application.

Thank you for your advice!

dorodnic, 2019-05-03:
Hi @evercoast-dev 
`context` is mainly used as `device` factory. It does not own device objects it helped to create (you can delete context as soon as you are done with it). It also, at least as far as I remember, is fairly trivially constructed, so in theory you could always create `context` on the stack.

dorodnic, 2019-05-03:
That said, `device` class does have some software state and every call to `query_devices` will result in a new `device` object instance. Hence, if you are accessing the device from different points in your application it might be a good idea to create it as soon as possible and share within your application.

evercoast-dev, 2019-05-03:
Excellent, thank you!

closed 2019-05-03 by evercoast-dev