realsenseai/librealsense #3892: Multi device connection problems

mmoyes, 2019-05-02:
| | |
|---|---|
|**librealsense**|2.21.0 RELEASE|
|**OS**|Linux|
Plugging in d435i and t265 struggles with: Worked previously with 2.19. Tried putting each device on different USBs (e.g. not both from front usb ports). All USBs are USB3.1

>>  02/05 12:26:35,066 ERROR [139996581316352] (types.h:243) get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy
 02/05 12:26:35,068 WARNING [139996581316352] (rs.cpp:279) null pointer passed for argument "device"

mmoyes, 2019-05-02:
Here's a hack I used to keep working...
```
	bool ready = false;
	while (!ready) {
		try {
		{
		    rs2::config cfg;
		    cfg.enable_stream(RS2_STREAM_COLOR, 1280,720, RS2_FORMAT_BGR8);
		    pipes[1].start(cfg);
		}
		{
		    rs2::config cfg;
		    cfg.enable_stream(RS2_STREAM_POSE, RS2_FORMAT_6DOF);
		    pipes[0].start(cfg);
		}
		ready = true;
		} catch (const rs2::error & e)
		{
			std::cerr << "RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
		}
	}
	std::cout << "\n\nAll Realsense devices loaded" << std::endl;
```

It spams with errors like;

> RealSense error calling rs2_pipeline_start_with_config(pipe:0x5558c07e3680, config:0x5558c08009b0):
    get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy

Till it decides to work

ev-mp, 2019-05-02:
@mmoyes hi, 
Version v2.19.2 introduced [certain changes](https://github.com/IntelRealSense/librealsense/pull/3515) to allow T265 seamless integration into the SDK initialization flow.
This resulted in imposing a constrain for all T265 devices to be mapped and accessed via with the first `context` object allocated in the code. https://github.com/IntelRealSense/librealsense/issues/3734#issuecomment-481982507

In the above case you should probably review the initialization of `pipes[0] pipes[1]` to make sure they receive the shared resource.

Refer to [`multicam`](https://github.com/IntelRealSense/librealsense/blob/master/examples/multicam/rs-multicam.cpp#L15-L29) example for the full code.

#3734, #3434, #3437

mmoyes, 2019-05-03:
@ev-mp the above was a hack that just keeps trying till it succeeds. The problem I am having is with multicam example and realsense-viewer

mmoyes, 2019-05-03:
Just checked on Windows and I have no issues. So maybe some kind of libusb issue?

I did not have this issue with earlier releases (where you had to query_devices() a couple of times to get the t265 to show up.

RealSenseCustomerSupport, 2019-05-17:
----------------------------------------------

@mmoyes - Is the issue only when you have a D435i connected in the system with T265?  So if you just have T265 in the system, you can use the T265 device and if you have D435i in the system only you can use that device without issue?

When you have both cameras connected and you do a query_devices call, what devices show up?  Does the T265 show up as a T265 or Movidius?

mmoyes, 2019-05-19:
Both devices work fine alone. The d435i works with d435is plugged in. Multiple t265s untested

ev-mp, 2019-05-20:
@mmoyes , do you manage to run `rs-multicam` with both devices connected ?
If so - what are the results.
Also provide the output of  `rs-enumerate-devices`

mmoyes, 2019-05-20:
@ev-mp
**rs-multicam** 
```
RealSense error calling rs2_pipeline_start_with_config(pipe:0x555bb4f97cc0, config:0x555bb5012b20):
    get_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy`
```

**rs-enumerate-devices**
Returns just the d435i. 
Opening realsense-viewer displays the d435i and gives you the option to add t265 from the "Add Source" menu

mmoyes, 2019-05-28:
@ev-mp 
Tried again with 2.22 and got the same results

ev-mp, 2019-05-29:
@mmoyes hello,
Please run `rs-multicam` with librealsense logs enabled:
- Add environmental variable `LRS_LOG_LEVEL=DEBUG`
- Run `rs-multicam`

mmoyes, 2019-05-29:
@ev-mp ,Same as before
> RealSense error calling rs2_pipeline_start_with_config(pipe:0x559d196951b0, config:0x559d1968e570):
    set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy

yyxr75, 2019-08-20:
Any progress on this? 
I just encountered the same problem.

yyxr75, 2019-08-21:
> @ev-mp ,Same as before
> 
> > RealSense error calling rs2_pipeline_start_with_config(pipe:0x559d196951b0, config:0x559d1968e570):
> > set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Device or resource busy

Any progress on this?
I just encountered the same problem.

RealSenseCustomerSupport, 2020-01-14:
----------------------------------------------

Hi mmoyes....Are you still seeing this issue with latest. If so can you please provide some details on how to reproduce.

Thanks

mmoyes, 2020-01-14:
Not looked in to it recently. I put code to replicate in thread here: https://github.com/IntelRealSense/librealsense/issues/3892#issuecomment-489170813

RealSenseSupport, 2020-09-09:
Thank you for highlighting the issue with multi device connections. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue with the T265.

closed 2020-09-16 by RealSenseSupport