realsenseai/librealsense #3893: T265 M3 screw holes loose

pkrush, 2019-05-02:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 

### Issue Description
Anyone else having issues with the two M3 screw holes on the T265? I have two of these T265 cameras and both holes only catch on the last thread, or not at all. I have tried a number of different M3(0.5 pitch) screws and they all act the same. I don't have an issue with the M3 holes on the D435 models.  Am I missing something? Thanks!

SlavikLiman, 2019-05-05:
Hi @pkrush ,
Just tested 3 x T265 cameras (total of 6 holes) and 1 x D435 camera. I marked the depth of the screw, when it's in the whole so I can compare how far the screw goes within the whole. For all 8 holes (3 x T265 + 1 x D435) it seems the same.

pkrush, 2019-05-05:
I'm sure it was just a bad part of the run.

closed 2019-05-05 by pkrush

eric-schleicher, 2019-05-06:
We tool have this issue.  Our customers have however indicated that they expected intel hardware to come connected with tape :(

ptyork, 2019-05-11:
For what it's worth, I too am experiencing this. D435i is perfect. T265 won't hold a screw at all. Definitely at least one run of these were badly tapped. Which is a real shame because I'm mounting mine to a human via a GoPro adapter, and obviously need a pretty solid connection. I'd prefer not to join Eric in resorting to duct tape.

PaulKrush, 2019-05-13:
We just got two new T265's and the these have good threads. It's duck tape for the old ones...

bfulkers-i, 2019-05-13:
@PaulKrush @ptyork @eric-schleicher, this does look like a bad run, if you'd like to get it fixed you can likely do a warranty exchange with the place you bought the T265s from.