realsenseai/librealsense #3894: How to display part of image to the pointcloud using RealSense

Resays, 2019-05-03:
Actually, I am working on something, related to pointcloud and I need to grab a depth image and display only the part of pic to the pointcloud as a 3D representation 
I successfully executed the pointcloud example with a live camera, but I need to get a pic and display its part to the point cloud.

Help ... how to accomplish the task?

Leo16112, 2019-05-03:
Hello,
you can create RGB Mat containing XYZ coordinates( there are examples for getting world coordinates), then just crop part of this XYZ Mat and create point cloud from it.

dorodnic, 2019-05-03:
Hi @Resays 
librealsense is not a point-cloud manipulation library, but there are some things you can do. If you need to cut based on depth, `threshold` processing-block can do that. More complex operations can be accomplished by dedicated libraries (PCL and Open3D are good candidates) or by manipulating the XYZ coordinates manually as @Leo16112 suggested.

RealSenseCustomerSupport, 2019-05-09:
----------------------------------------------

@Resays Any other questions about this ticket? Looking forward to your update. Thanks!

Resays, 2019-05-13:
Here is the problem, let's say I have to capture the image with realsense and display the image in 3D after object detection using YOLO. so YOLO gave me the image with identification of bounding box and now I have to display in 3D whatever inside the box.
an example image is attached.
[image]

Resays, 2019-05-13:
I'm not getting the point, how will I crop the image and display in 3D, opencv can crop the jpeg image but i need to display it in 3D, so i think I need depth data for what is inside the bounding box

Resays, 2019-05-13:
> Hi @Resays
> librealsense is not a point-cloud manipulation library, but there are some things you can do. If you need to cut based on depth, `threshold` processing-block can do that. More complex [...]

And how can i get xyz coordinates?
i used your code to get depth data and RGB data but not getting the point how to get x,y,z coordinates please guide me.

Resays, 2019-05-13:
> Hello,
> you can create RGB Mat containing XYZ coordinates( there are examples for getting world coordinates), then just crop part of this XYZ Mat and create point cloud from it.

can you refer any example, how to get xyz coordinates and how to crop image for displaying in 3D ?

Resays, 2019-05-13:
[image]
[image]
I successully captured these image using c++ and realsense, using OpenCV

```
	cv::Mat color(Size(640, 480), CV_8UC3, (void*)color_frame.get_data(), Mat::AUTO_STEP);
	cv::Mat ir(Size(640, 480), CV_8UC1, (void*)depth_frame.get_data(), Mat::AUTO_STEP);
```

whats next how to get xyz data from RGB Mat images.

Leo16112, 2019-05-13:
> > Hello,
> > you can create RGB Mat containing XYZ coordinates( there are examples for getting world coordinates), then just crop part of this XYZ Mat and create point cloud from it.
> 
> can you refer any example, how to get xyz coordinates and how to crop image for displaying in 3D ?

Hello,
by the way, i have to say, that this point should be explained better. I also spend a lot of time to solve this task and started already hate realsense, because in case of Kinect and OpenNI it is just one line of code. And here at those moment i spend a lot of time to solve the task. It would be very nice to have something like "rs2::frame xyz = data.get_xyz_frame();", or just some function for converting. But i solved the task for me and i love realsesense again :D

So, you can find in this example the transform function.
https://github.com/IntelRealSense/librealsense/tree/master/examples/measure
Then you need create your own, that goes trough all pixels and converts them to 3d, where you have XYZ in BGR(or RGB, depends on you) channels. But pay attention, that you get intrinsics outside the loops. Then you can just crop ROI and transform to type you need to provide for viewer.

Resays, 2019-05-13:
Thanks for your reply, have you reviewed my reply with images a little upside on this post?
and am I going in the right direction?
and where is an image in your example? or maybe I didn't get how to run this example

RealSenseCustomerSupport, 2019-06-24:
----------------------------------------------

@Resays Sorry for the late response! Did you get it through now?

Resays, 2019-06-24:
Hi, Yes I got thembut the details about working with realsense is too weak ... and i having a hard time to grab the information about realsense 

Thanks anyway .. if i need more information i'll contact you again
Siddiqui, Muhammad YasirPh.D. Scholar, Tongmyong University

RealSenseCustomerSupport, 2019-06-26:
----------------------------------------------

@Resays Glad to know this resolved. Not sure if you have noticed our official site https://www.intelrealsense.com/  Please check there to see if you can find what you want. And if still can't get answer, please feel free to open tickets here. Thanks!

closed 2019-07-01 by RealSenseCustomerSupport

fburak, 2020-03-02:
> 
> 
> Here is the problem, let's say I have to capture the image with realsense and display the image in 3D after object detection using YOLO. so YOLO gave me the image with identification of bounding box and now I have to display in 3D whatever inside the box.
> an example image is attached.
> [image]

Almost same problem for me. I have to get point cloud of an object in depth frame by using bounding boxes generated from rgb image with an image processing algorithm.

how did you solved? Is is possible to share function you wrote? @Resays 
or is there an automated way to do this(cropping depth frame and generating point cloud from it) in new releases? @dorodnic