realsenseai/librealsense #3895: Errors installating Intel RealSense on Ubuntu

henrikcheung, 2019-05-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 18.04)  }  | 
| Kernel Version (Linux Only)    |  (4.15.0)        | 
| Platform                                 | Raspberry Pi  |
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
Hi all, I install Intel Realsense on Ubuntu following this: https://github.com/IntelRealSense/librealsense/blob/v2.20.0/doc/distribution_linux.md 

And two problems are found that I couldn't install it.
1. During the second command,
"Add the server to the list of repositories (Ubuntu both 16 and 18 LTS):
sudo add-apt-repository "deb http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial main" -u"

The following message came up:
"N: Skipping acquire of configured file 'main/binary-armhf/Packages' as repository 'http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo xenial InRelease' doesn't support architecture 'armhf'"

2. During the third command,
"Install the libraries (see section below if upgrading packages):
sudo apt-get install librealsense2-dkms"

The following message came up:
"E: Unable to locate package librealsense2-dkms"
 
I have tried them with and without the camera connected, have tried to update "sudo apt-get update" and upgrade by "sudo apt-get upgrade". But still got the two error messages above afterwards.

What should I test or try please? Thanks!

ev-mp, 2019-05-03:
@henrikcheung , the debian packages are available for X86/AMD64 CPUs.

You can find  Raspberry-dedicated guides in the following links:
- https://github.com/IntelRealSense/librealsense/blob/v2.20.0/doc/installation_raspbian.md
- https://github.com/IntelRealSense/librealsense/blob/v2.20.0/doc/RaspberryPi3.md

henrikcheung, 2019-05-06:
Thanks @ev-mp , I'm now following your first link to install. While installing the packages, the following messages showed up:
E: Unable to locate package libdrm-amdpu1-dbg
E: Unable to locate package libdrm-exynos1-dbg
E: Unable to locate package libdrm-freedreno1-dbg
E: Unable to locate package libdrm-nouveau2-dbg
E: Unable to locate package libdrm-omap1-dbg
E: Unable to locate package libdrm-radeon1-dbg
E: Unable to locate package libdrm-tegra0-dbg
E: Unable to locate package libdrm2-dbg

May I know am I using outdated codes? (I will continue installing other steps the same time, thanks!)

closed 2019-05-07 by henrikcheung

GOBish, 2019-10-01:
I get the same errors, were you able to solve this?