realsenseai/librealsense #3896: Realsense won't release file handles to bag-files in unity *build*

Dealerpriest, 2019-05-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D415 | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |  Win 10  | 
| Platform                                 | PC |
| SDK Version                            | build 2.21.0                          | 
| Language                            |  C#/unity                          | 
| Segment			| VR   | 

### Issue Description
A project that works fine in editor, behaves completely differently when built to executable.

I have a Unity project where I can successfully record from 2 realsense cameras to bag-files and directly play them back. I have made my own custom script that retrieves all connected cameras and initializes pipes using config objects. When recording should stop I `Stop()` the pipes and `Dispose()` them. Then create new pipes with playback configuration (also using a config object). It plays back fine without problem. In the unity editor, that is.

If.
I build the unity project to windows standalone (.exe) the program fails. Specifically, it fails because the recording pipes won't let go of the files when the recording is stopped. Hence, when the playback is to be initialized I get exceptions of the filepath being invalid. I'm rather sure it's because the program somehow keeps the file handles even after the recording is stopped. I can monitor the file structure in windows explorer while running the program. The program successfully creates a folder and two bag files in it when the recording is started. explorer shows the file size of both files to be 0 bytes. It stays this way even when recording is stopped. When trying to start playback the program semi crashes (playback won't start and a few exceptions are raised). The file size still showing 0 bytes. If quitting the program the file sizes immediately changes to reflect all bytes recorded into the files.

This can be compared to when I run the project in the editor. As soon as recording is stopped this is reflected in windows explorer and the file sizes go from 0 bytes to a few hundred. I will try to provide a minimal script (monobehaviour) to reproduce this issue.

Dealerpriest, 2019-05-03:
Here is a minimal scirpt to reproduce the bug. The script uses keyboard input to switch between streaming, recording & playback. So one could start a scene with this script on a gameobject. Then press **R** to start recording. Then **P** to start playback. This all works fine in the editor. But not when built as a windows standalone application. The script implements `RsFrameProvider` so it's possible to connect the script to `RsProcessingPipe` or such for visualization.
@ogoshen. Perhaps you have an idea what could be the issue?
```
using System.Collections;
using System.Collections.Generic;
using UnityEngine;
using Intel.RealSense;
using System.IO;
using System;

public class SimpleRecorder : RsFrameProvider {
	private SimpleRecorder self;
	bool streaming = false;
	Pipeline pipe;

	string recordPath;

	//RsFrameProvider stuff
	public override event Action<PipelineProfile> OnStart;
	public override event Action OnStop;
	public override event Action<Frame> OnNewSample;

	void Awake() {
		self = this;

		
	}

	void Start() {
		string baseDir = Directory.GetParent(Application.dataPath).FullName;
		string pathToAllRecordings = Path.Combine(baseDir, "recordings");

		if (!Directory.Exists(pathToAllRecordings)) {
			Directory.CreateDirectory(pathToAllRecordings);
		}

		recordPath = Path.Combine(pathToAllRecordings, "recording.bag");

		pipe = new Pipeline();
		pipe.Start();
	}

	void Update() {
		if (Input.GetKeyUp(KeyCode.R)) {
			pipe.Stop();
			pipe.Dispose();
			Config cfg = new Config();
			cfg.EnableRecordToFile(recordPath);
			pipe = new Pipeline();
			pipe.Start(cfg);
		}
		if (Input.GetKeyUp(KeyCode.S)) {
			pipe.Stop();
			pipe.Dispose();
			pipe = new Pipeline();
			pipe.Start();
		}
		if (Input.GetKeyUp(KeyCode.P)) {
			pipe.Stop();
			pipe.Dispose();
			Config cfg = new Config();
			cfg.EnableDeviceFromFile(recordPath);
			pipe = new Pipeline();
			pipe.Start(cfg);
		}

		if (pipe.PollForFrames(out FrameSet frames)) {
			using (frames) {
				if (!streaming) {
					OnStart?.Invoke(pipe.ActiveProfile);
					streaming = true;
				}

				OnNewSample(frames);
			}
		}

	}

	private void OnDestroy() {
		if (OnStop != null)
			OnStop();
	}
}
```

Dealerpriest, 2019-05-03:
In my efforts to debug this I have now come to the conclusion that the problem must be related to some reference to the recording pipe/device not getting cleaned up/released/disposed . I tried throwing in a call to `GC.Collect();` directly after each call to `pipe.Dispose()` and it immedately helped in releasing the "lock" of the bag file.
But this can't be the expected way to handle this?! Is there some other measures I'll have to take to truly release the pipe when I want to replace it with a `new PipeLine()`?

dorodnic, 2019-05-03:
Hi @Dealerpriest 
If `GC.Collect` helps, there must be some missing `Dispose` hiding somewhere. I'm not entirely sure where. @ogoshen perhaps you have an idea?

ogoshen, 2019-05-06:
@dorodnic is right, `GC.Collect` is a code smell for unreleased resources.

When running into issues like this, I recommend going into strict disposal mode.
For instance, `Config` objects are also `IDisposable` and are very short lived, yet you're not explicitly releasing them, a `using` statement would both keep their scope tight and release them immediately.

On to the issue at hand, I haven't tested the code but if i had to guess, I'd suspect it's actually the `PipelineProfile`  object here: 
```cs
OnStart?.Invoke(pipe.ActiveProfile);
``` 
that's keeping the `Pipeline` alive, and so the recorded file isn't properly flushed.
You might want to keep a reference to the `pipe.ActiveProfile` instance, disposing it before the pipeline.

Dealerpriest, 2019-05-06:
@ogoshen, I tried what you said. Unfortunately it didn't resolve the issue. I attach my current code. As of now, I can't find any other resource to dispose? Have I missed something?
 To be clear, this works in the unity editor but not when built out to a windows executable. To be even more clear. It does works if you uncomment the `GC.Collect()` calls.

@ogoshen would it be possible for you to try the script? I was careful to make it compatible with the realsense prefabs by implementing the `RsFrameProvider` interface, so it should be rather easy for you to just drop the script into a realsense unity project.

-------------------
```
using System.Collections;
using System.Collections.Generic;
using UnityEngine;
using Intel.RealSense;
using System.IO;
using System;

public class SimpleRecorder : RsFrameProvider {
	private SimpleRecorder self;
	bool streaming = false;
	Pipeline pipe;
	PipelineProfile activeProfile;

	string recordPath;

	//RsFrameProvider stuff
	public override event Action<PipelineProfile> OnStart;
	public override event Action OnStop;
	public override event Action<Frame> OnNewSample;

	void Awake() {
		self = this;

		
	}

	void Start() {
		string baseDir = Directory.GetParent(Application.dataPath).FullName;
		string pathToAllRecordings = Path.Combine(baseDir, "recordings");

		if (!Directory.Exists(pathToAllRecordings)) {
			Directory.CreateDirectory(pathToAllRecordings);
		}

		recordPath = Path.Combine(pathToAllRecordings, "recording.bag");

		pipe = new Pipeline();
		pipe.Start();
		activeProfile = pipe.ActiveProfile;
	}

	void Update() {
		if (Input.GetKeyUp(KeyCode.R)) {
			activeProfile.Dispose();
			pipe.Stop();
			pipe.Dispose();
			//pipe = null;
			//GC.Collect();
			using (Config cfg = new Config()) {
				cfg.EnableRecordToFile(recordPath);
				pipe = new Pipeline();
				pipe.Start(cfg);
			}
			activeProfile = pipe.ActiveProfile;
		}
		if (Input.GetKeyUp(KeyCode.S)) {
			activeProfile.Dispose();
			pipe.Stop();
			pipe.Dispose();
			//pipe = null;
			//GC.Collect();
			pipe = new Pipeline();
			pipe.Start();
			activeProfile = pipe.ActiveProfile;
		}
		if (Input.GetKeyUp(KeyCode.P)) {
			activeProfile.Dispose();
			pipe.Stop();
			pipe.Dispose();
			//pipe = null;
			//GC.Collect();
			using (Config cfg = new Config()) {
				cfg.EnableDeviceFromFile(recordPath);
				pipe = new Pipeline();
				pipe.Start(cfg);
			}
			activeProfile = pipe.ActiveProfile;
		}

		if (pipe.PollForFrames(out FrameSet frames)) {
			using (frames) {
				if (!streaming) {
					OnStart?.Invoke(activeProfile);
					streaming = true;
				}

				OnNewSample(frames);
			}
		}

	}

	private void OnDestroy() {
		if (OnStop != null)
			OnStop();
	}
}
```

RealSenseCustomerSupport, 2019-08-26:
----------------------------------------------

@Dealerpriest Sorry for late response. Is the issue still there? Looking forward to your reply. Thanks!

ogoshen, 2019-08-28:
> @ogoshen, I tried what you said. Unfortunately it didn't resolve the issue. I attach my current code. As of now, I can't find any other resource to dispose? Have I missed something?

Yes you have, and your intuition was correct. The `PipelineProfile` returned from `Pipline.Start` is what's keeping things alive too long, hence `GC.Collect` resolved your issues.

If the object is otherwise unused just replace the `Start` calls with:
```cs
pipe.Start(cfg).Dispose();
```

As to why this didn't happen in the editor, I can only assume the editor is generating some mild garbage and was collecting when the sdk added pressure.

Sorry this took so long...

closed 2019-08-28 by Dealerpriest