realsenseai/librealsense #3897: Timestamp reports wrong time!!!

Dealerpriest, 2019-05-03:
OMG!
Is this for real!!
The SDK documentation clearly states that the timestamp associated with each frame should be the duration since the device was turned on (https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1frame.html#a25f71d45193f2f4d77960320276b83f1).
Yes. I've followed the steps to make the device report "hardware clock" (and verified it does)  as per these instructions:
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md#enabling-metadata-on-windows

How in gods name then, is it possible for me to make the clock lag, by doing various calls to the SDK (through the C# wrapper) where I create and dispose pipes with different configuration?

I am using the timestamps to align frames from different cameras. I was confused that the syncronisation sometimes was off. So I just now made an experiment where I use a stopwatch and continuously monitor the timestamps I receive from the incoming frames metadata.
I then stop the pipes, dispose them and then create new ones. For each time I do I can clearly see how the timestamps reported from the incoming frames lags behind the stopwatch. To me this makes no sense!

I might come across as somewhat upset. To be honest I am. I feel this way of operation is so far from what is stated that it is borderline ridiculous.

I would be grateful for any advice!! All I'm trying to do is to create syncronized recording/playback. I've been trying at this for almost two months now and I continuously encounter weird behaviour, forcing me to program "around" the problem. How hard can it be?! To me it feels like a VERY likely use case!

dorodnic, 2019-05-03:
Hi @Dealerpriest 
Understand your frustration, but lets try to be constructive about it. 
Hardware clocks are known to drift. This is why [multicamera white paper](https://dev.intelrealsense.com/docs/multiple-depth-cameras-configuration) is recommending to bundle frames based on time of arrival.
What needs to be done is to programmatically keep an eye on each cameras clock and adjust everything accordingly. We have been working on this for some time, but it is not quite ready for a pull-request yet (very close though) All our development is in the open, so you can review and try it if you'd like - [diff](https://github.com/IntelRealSense/librealsense/compare/development...doronhi:hwsystime) and the pull-command should be:
```sh
git checkout -b doronhi-development hwsystime
git pull https://github.com/doronhi/librealsense.git hwsystime
```
There are many subtle corner cases that need to be addressed, but once all done, this will allow to bundle frames from different devices and different types of devices) using `syncher` object.

jbohnslav, 2019-05-03:
Looking forward to a nice `syncher` object that handles some of these details!

Dealerpriest, 2019-05-05:
@dorodnic, thanks for the swift reply! I appreciate the help. I want to be clear that to me this isn't what I would consider drift. It's more like the hardware clocks of the devices are stopped/paused for a short duration. I can try to provide a c#/unity script to reproduce. Would that be of help?

dorodnic, 2019-05-05:
@Dealerpriest 
If you can, please share the code to make sure we are talking about the same problem.
Another behavior that I think can occur with hardware clocks is that it can be restarted due to change in FPS because of auto-exposure or setting a JSON preset. 
The PR in question should be able to handle that case as well.

RealSenseCustomerSupport, 2019-06-08:
----------------------------------------------

Hi @Dealerpriest,
The ticket has been idle for a while and I am just wondering whether or not you have found the solution for this issue already?
Thank you.

Dealerpriest, 2019-06-11:
Unfortunately not. I have provided code above to easily reproduce the bug. I think it would be of interest for an Intel developer (perhaps @dorodnic or @ogoshen) to try it out. Then you could verify that this is a bug and deal with it accordingly.

Dealerpriest, 2019-08-12:
@dorodnic or @ogoshen or @RealSenseCustomerSupport. Have you had the possibility to verify this bug by trying out the code I provided above?

RealSenseCustomerSupport, 2019-08-13:
----------------------------------------------

Only to help to debug, I tried to simplify the test using the following code. It will print out the clock domain, back-end timestamp (system host time) and the frame timestamp (hardware device time) from the frame metadata. The pipe will restarts every 5 seconds.
The result of two timestamps has been consistent after restarting the pipeline.

        config.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);

        while (1)
        {
            pipeline.start(config);

            auto last = std::chrono::high_resolution_clock::now();
            auto now = std::chrono::high_resolution_clock::now();
            while (now - last < std::chrono::seconds(5))
            {
                if (!pipeline.try_wait_for_frames(&frameset, 100)) {
                    fprintf(stderr, "Wait for frames timed out\n");
                    continue;
                }

                for (auto&& frame : frameset) {

                        auto ts_bkend = frame.get_frame_metadata(RS2_FRAME_METADATA_BACKEND_TIMESTAMP);
                        auto ts_frame = frame.get_frame_metadata(RS2_FRAME_METADATA_FRAME_TIMESTAMP);
                        auto ts_domain = frame.get_frame_timestamp_domain();
                        auto frm_id = frame.get_frame_number();
                        const char* ts_domain_str = rs2_timestamp_domain_to_string(ts_domain);
                        printf("%s, %lld, ts_bkend:%lld, ts_frame:%lld, diff:%lld \n", ts_domain_str, frm_id, ts_bkend, ts_frame, (ts_bkend - ts_frame/1000));
                    }
                    now = std::chrono::high_resolution_clock::now();
            }
            pipeline.stop();
        }

RealSenseCustomerSupport, 2019-08-23:
----------------------------------------------

Does the device report consistent timestamps with the code or could we close this now?

Dealerpriest, 2019-08-30:
I will try the code you provided asap and report back my findings👍

RealSenseCustomerSupport, 2019-09-20:
----------------------------------------------

Hi @Dealerpriest,
Please kindly let us know if you have any question. Thank you.

closed 2019-10-11 by RealSenseCustomerSupport