realsenseai/librealsense #3898: Change to color exposure between 5.10.3 & 5.11.6

evercoast-dev, 2019-05-03:
Camera Model: D400
Firmware Version: 5.11.6
Operating System: Windows & Ubuntu
Kernel Version: 4.15.0-48-generic
Platform: PC
SDK Version: 2.16.0 - 2.20.0
Language: C++
Segment: VR/AR

Hi all,

We noted a considerable change to the color exposure between firmware versions 5.10.3 and 5.11.6. For clarity, two questions:

What actually changed and what was the purpose?

Were there other related changes, to infrared exposure for instance that may be less visible?

Thank you!

RealSenseCustomerSupport, 2019-05-09:
----------------------------------------------

@evercoast-dev Could you please provide some snapshots to clarify what's the detailed difference you've seen between these two FW? Thanks!

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

@evercoast-dev Could you please provide more details? Thanks!

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@evercoast-dev Do you still need support on this? Please update. Thanks!

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@evercoast-dev Sorry we didn't get response from you for weeks. Will close the ticket if no other questions.

closed 2019-06-03 by RealSenseCustomerSupport