realsenseai/librealsense #3899: Hardware sync pulse duration is very short

jbohnslav, 2019-05-03:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | v5.11.1.100  | 
| Operating System & Version |  Windows 10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC |
| SDK Version                            |  2.17.0                     | 
| Language                            |  python                    | 
| Segment			|  other                  | 

### Issue Description
I'm currently using the realsense in a scientific application, where it is required to have a voltage signal when frames were acquired for synchronization to other devices. I'm copying the master signal into my data acquisition board. The cameras are running at 60 Hz, so in theory it should be very easy to detect a pulse train at 60 Hz. However, the sync pulse duration is incredibly short--only 50 microseconds! This is a 20 kHz signal, much larger than the framerate. To stay on the right side of Nyquist, we should then sample at 40kHz or more, which is vastly higher than necessary for any of our other devices. This leads to more CPU usage and much larger file sizes. 

Is there any way to increase the duration of the sync signal? Could this be changed in firmware? Point Grey cameras (for example) have a ~1ms strobe signal output, or can be set to be exactly the exposure time of the camera, which is very useful.

RealSenseCustomerSupport, 2019-05-08:
----------------------------------------------

Actually the pulse width initially was set at 300ns, and we extended it to 50us around 1.5 years ago.
Due to internal counter limitation, the max pulse width can reach 200us. Is this good for you?

Thanks

jbohnslav, 2019-05-10:
Hi there! Thanks for your response. 200 us would be helpful, as we could get more samples or reduce our sampling rate. How do I do this?

RealSenseCustomerSupport, 2019-05-10:
----------------------------------------------

I am sorry. I just realized that your camera is locked by factory which means it's blocked to make any change to register settings, including the one for the pulse width adjustment. I will submit a Change Request to development team for this request.

Thanks

closed 2019-06-10 by RealSenseCustomerSupport