realsenseai/librealsense #3903: Realsense SR300 does not work on Jetson TX2

zmnh2003, 2019-05-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | SR300 | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Linux(Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  4.4.38-tegra                               | 
| Platform                                 | NVIDIA Jetson TX2 |
| SDK Version                            |  2.21                        | 
| Language                            |  python                          | 
| Segment			|  Robot                          | 

### Issue Description
I installed pyrealsense2 following #2586 issue. I believe its installed well.
And this is a example code which i tried.

import pyrealsense2 as rs
import numpy as np
import cv2

# Configure IR stream
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.infrared, 640, 480, rs.format.y8, 30)

# Start input streaming
pipeline.start(config)

# Ignore first 1sec for camera warm-up
for i in range(30):
    frames = pipeline.wait_for_frames()

try:
    while True:

        # Read image
        frames = pipeline.wait_for_frames()
        infrared_frame = frames.first(rs.stream.infrared)
        IR_image = np.asanyarray(infrared_frame.get_data())

        # Display image
        cv2.imshow('IR image', IR_image)

        # Exit on ESC key
        c = cv2.waitKey(1) % 0x100
        if c == 27:
            break

finally:
    pipeline.stop() 
    cv2.destroyAllWindows()

I thought it should be done. But it made error like 

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: libusb_open(...) returned LIBUSB_ERROR_ACCESS Last Error: Permission denied

i found many solutions including 
/etc/udev/rules.d/90-usbpermission.rules << this one and
sudo adduser "username" plugdev << this one.

But in my TX2 /etc/udev directory doesn't exists and sudo adduser didn't make any sense.
PLEASE HELP ME

RealSenseCustomerSupport, 2019-05-08:
----------------------------------------------

Hi @zmhn2003,

Please also check if you follow below instruction to setup the system environment.
https://github.com/jetsonhacks/buildLibrealsense2TX
Make sure to apply udev rules and reset your platform:
$ sudo cp config/99-realsense-libusb.rules /etc/udev/rules.d/
$ sudo udevadm control --reload-rules && udevadm trigger

zmnh2003, 2019-05-09:
Hi,

Thanks for your kindness. I treid many ways and it works now. Thank you!

closed 2019-05-09 by zmnh2003