realsenseai/librealsense #3904: subpixel in the formula of depth RMS error

Medo91code, 2019-05-04:
When i read the realsense Documentations , i have a confusing about the subpixel .
How can i define or find a subpixel in the formula of depth RMS error .
i can get the subpixel RMS error in Depth Quality Tool.
is the subpixel in the formula of depth RMS error equal to the subpixel RMS error in Depth Quality Tool?

[image]
[image]

agrunnet, 2019-05-04:
The subpixel is essentially an empirical number. For a well calibrated RealSense camera it could in principle be as low as 0.032 but it is best to use about 0.08. That is the short answer. 

The longer answer is that “it depends”. The stereo algorithm can produce depth with subpixel of 0.032 for a well textured scene but there are other limitations that come in: 1. How well textured is the scene? 2. If using laser projectors they tend to add noise (see the white paper on projectors) 3. There are calibration and lens distortions that can limit whole FOV uniformity so it is best to characterize across a small FOV of say 10-20%.

Medo91code, 2019-05-04:
@agrunnet  thank you for your answer 
I want to know ...
is the subpixel in the formula of depth RMS error equal to the subpixel RMS error in Depth Quality Tool?

agrunnet, 2019-05-04:
Yes. 
But please note that to make sure you are measuring the subpixel behavior of the camera (and not lens distortion or object flatness) it is important to 1. Look at a small ROI and 2. Set the depth units to 100. By default it is 1000 which means that it outputs depth in 1mm units. 100 will output depth in 100um units. This is also described more in the projector white paper.

closed 2019-06-19 by RealSenseCustomerSupport