realsenseai/librealsense #3905: "Unsupported GCC version" error during Odroid XU4 build

kvbogdanov, 2019-05-04:
Trying to build v. 2.21.0 for Odroid XU4 (kernel 4.9.27-35)

Got an "Unsupported GCC version" with two different GCC versions - 4.8.5 and 5.3.0 in file src/ds5/advanced_mode/json_loader.hpp that results in complete build fail.

In the error message, they suggest opening https://github.com/nlohmann/json there supported compiler listed, and 4.8.5 in the list (I know that I should use gcc 5 instead of gcc 4, but anyway 4.8.5 should not bring this error)

RealSenseCustomerSupport, 2019-05-09:
----------------------------------------------

@kvbogdanov Could you please try GCC 5 to see if still have issue? Thanks!

kvbogdanov, 2019-05-09:
@RealSenseCustomerSupport yes, I've already tried GCC 5.3.0 and gets the same error (but at least no warnings in opposite to 4.8.5)

RealSenseCustomerSupport, 2019-05-15:
----------------------------------------------

@kvbogdanov The default GCC version on my Odroid XU4 is 5.4.0 and I didn't get the error you reported.  Did you install both gcc 4.8.5 and 5.3.0? If so, please select gcc-5 by typing "sudo update-alternatives --config gcc" to try again. Thanks!

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@kvbogdanov Did you get it through? Could you please update? Thanks!

kvbogdanov, 2019-05-21:
@RealSenseCustomerSupport No, it didn't ( I tried also 5.4.0 and get the same:
[image]
[image]

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@kvbogdanov Did you run cmake again after you switched to GCC v5.4?

kvbogdanov, 2019-06-11:
@RealSenseCustomerSupport when I run cmake again, I get this error even earlier:
[image]

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@kvbogdanov That's wired. Did you run cmake in the previous build folder or another new build folder?

kvbogdanov, 2019-06-13:
@RealSenseCustomerSupport in the new folder

RealSenseCustomerSupport, 2019-06-14:
----------------------------------------------

@kvbogdanov Seems that gcc-5 still not used when build. Could you please provide the result of "update-alternatives --query gcc" and "update-alternatives --query g++"? Thanks!

kvbogdanov, 2019-06-18:
@RealSenseCustomerSupport sure, here it is:
[image]

RealSenseCustomerSupport, 2019-06-19:
----------------------------------------------

@kvbogdanov Seen from the result, g++ version is v4.8. Could you please switch gcc and g++ both to v5.0 to try again?

kvbogdanov, 2019-06-19:
@RealSenseCustomerSupport of course, I am going to do it ASAP, but I thought this code gets only GCC version info, not G++ (unlike __GNUG__ for example):

`#define GCC_VERSION (__GNUC__ * 10000 + __GNUC_MINOR__ * 100 + __GNUC_PATCHLEVEL__)`

RealSenseCustomerSupport, 2019-06-21:
----------------------------------------------

@kvbogdanov Any luck after set g++ version?

kvbogdanov, 2019-06-21:
@RealSenseCustomerSupport yes, that's it! Finally, It was my fault. Thanks for the support!

closed 2019-06-21 by kvbogdanov