realsenseai/librealsense #3906: Depth stream intermittently freezes for multiple seconds

WasabiFan, 2019-05-04:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.11.01.100 | 
| Operating System & Version |   Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  4.9 with patches from https://github.com/jetsonhacks/buildLibrealsense2Xavier                                         | 
| Platform                                 | NVIDIA Jetson Xavier  |
| SDK Version                            |  2.17.1 with patches from https://github.com/jetsonhacks/buildLibrealsense2Xavier                           | 
| Language                            |  Python                          | 
| Segment			|  Robotics                          | 

### Issue Description

The depth stream intermittently stops updating (continuously returns duplicate frames) for multiple seconds.

When that happens, messages such as the following begin to appear in the device log:

```
 04/05 13:44:30,109 DEBUG [546484228592] (sync.cpp:551) clean inactive stream in (TS: (FN: I Depth I Infrared I Infrared ) I Color )Depth Infrared Infrared 
```

and these stop appearing:

```
 04/05 13:44:30,076 DEBUG [546484228592] (sync.cpp:334) (TS: (FN: I Depth I Infrared I Infrared ) I Color ) Color 10250 358163.091000  Wait for missing stream: 7 next expected 358173.2523328 next expected 358173.2523329 next expected 358173.252332
```

I've been unable to find a root cause.

At these times, the color stream continues to update; however there are cases where the same happens but in reverse (the color stream freezes but depth stream continues).

In the following log, the freezing began at `2019-05-04 13:44:30.119630` and continued until I killed the app (just after `2019-05-04 13:44:32.546974`).

[2019-05-04-13_38_29.log](https://github.com/IntelRealSense/librealsense/files/3144828/2019-05-04-13_38_29.log)

I would appreciate suggestions from the development team! I'm comfortable debugging things if I can get some pointers.

RealSenseCustomerSupport, 2019-05-15:
----------------------------------------------

Hi @WasabiFan,
Do you find the issue dependent on the frame rate, resolutions and the number of enabled streams?
I attempt to replicate the problem by running Z16+Left IR +Right IR+RGB, all at 1280x720 at 30fps but have not experienced the stream stop yet.
How is the D435(s) connected to the Xavier? Is it directly attached to TypeA/C port or via USB hub or via other types of connection? What's the length of the USB cable?

WasabiFan, 2019-05-16:
@RealSenseCustomerSupport It is connected directly. This happens both with the USB C-to-A cable provided with the RealSense and with my own USB C-to-C cables.

I can see if I can get a more detailed repro setup for you.

Are you testing on an actual hardware Xavier devkit?

RealSenseCustomerSupport, 2019-05-16:
----------------------------------------------

Hi @WasabiFan,

> Are you testing on an actual hardware Xavier devkit?

Yes. My test was performed based on NVidia Jetson AGX Xavier Developer Kit with L4T 31.1 and with patches from https://github.com/jetsonhacks/buildLibrealsense2Xavier.

RealSenseCustomerSupport, 2019-06-20:
----------------------------------------------

Ticket being closed due to inactivity for 30+ days.

closed 2019-06-21 by RealSenseCustomerSupport