realsenseai/librealsense #3908: Unregistered Media formats : [ UYVY ]; Supported: [ GREY UYVY Y12I Y8I  Z16  ]

Myzhencai, 2019-05-05:
| | |
|---|---|
[image]

|**librealsense**|2.21.0 RELEASE|
|**OS**|Linux|
|**Name**|Intel RealSense D435|
|**Serial Number**|827312072680|
|**Firmware Version**|05.11.04.00|
|**Advanced Mode**|YES|
|**Usb Type Descriptor**|3.2|

Please provide a description of the problem

ev-mp, 2019-05-05:
@Myzhencai hello,
The log is is informative - to highlight that `UYVY` stream format is not exposed for D435 model. The format stands for "Reconstructed RGB" stream type and is available for D410/D415 sensors only.

haxiomic, 2019-05-05:
Same warning for me on MacOS but I get a crash `Segmentation fault: 11` whenever I enable depth stream on realsense-viewer

It was working earlier in the day but now I get a crash 100% of the time

haxiomic, 2019-05-05:
Viewer doesn't crash if I change to 848x480 @ 90Hz before enabling

Myzhencai, 2019-05-06:
@ev-mp thanks for help, have a nice day :-).

closed 2019-05-12 by dorodnic