realsenseai/librealsense #3911: T265 and obstacle localization?

TDVL, 2019-05-05:
Hello. 

I saw the datasheet of the T265 camera? How can the camera localize the obstacles, for example on the wall? Is there a demo for this? Where can I find whitepapers about T265? Is it possible to combine the T265 with D415 to find obstacles, I mean to join the depth data somehow with T265?

Thank you.

Best regards

SlavikLiman, 2019-05-06:
Hi @TDVL ,
Following [this ](https://github.com/IntelRealSense/librealsense/issues/3806) thread will answer some of your questions.

SlavikLiman, 2019-05-07:
You might be also interested in [this ](https://github.com/IntelRealSense/realsense-ros/tree/occupancy-mapping/occupancy) occupancy map example.

TDVL, 2019-05-10:
Hello Slavik,
Thank you for your answer, I will check the thread you that you suggested.
KInd regards

closed 2019-06-17 by RealSenseCustomerSupport