realsenseai/librealsense #3913: ply file to depth map

Woodstock94, 2019-05-06:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Operating System & Version |   {Win (8.1/10)   |                                
| Language                               |  {python }             | 

### Issue Description
hi everyone, I'm working with pyrealsense2, my aim is to retrieve a depthmap by using only camera parameters and my ply file.
By following https://github.com/IntelRealSense/librealsense/issues/2204 I understand that is possible to use `color_pixel = rs.rs2_project_point_to_pixel(color_intrin, color_point)`.
This function will project all my 3d points to 2d. It is clear how to utilize it for a single point, but how can I then reconstruct an image given all the outputs of this function? I don't know how to deal with this. 
Thanks for attention

Leo16112, 2019-05-09:
I never did it, but i would start with example for showing of pointcloud with opencv-viewer. You need to do  projection of 3D object to 2D plane. It actually done there. But instead of color you need encode distance to point in meter and put it in according pixel.

RealSenseCustomerSupport, 2019-05-13:
----------------------------------------------

Hi Woodstock94,

If not yet, you can refer to this one for the idea of getting 2D from 3D.
https://github.com/IntelRealSense/librealsense/issues/1601

Thanks!

RealSenseCustomerSupport, 2019-05-22:
----------------------------------------------

Hi Woodstock94,

Wonder if there is any update on this one?
Or if nothing else is needed, this will be closed.

Thanks!

closed 2019-05-28 by RealSenseCustomerSupport