realsenseai/librealsense #3915: saving PLY files by using points.export_to_ply cost many time?

yuzihe, 2019-05-06:
i am using D415 to save PLY files,here is my code
        pointcloud pc;
        pc.map_to(colorFrames);
        auto points = pc.calculate(depthFrames);
        sprintf(filename,"%s%s%s","D:\\",c,"\\right_ply.ply");
        points.export_to_ply(filename,colorFrames);
but this code will cost 5-6s, anyone can help me to shorten the time?

RealSenseCustomerSupport, 2019-05-14:
----------------------------------------------

@yuzihe Did you get the issue with Debug build or Release build? Could you please try Release build which should be faster than Debug? Thanks!

RealSenseCustomerSupport, 2019-05-21:
----------------------------------------------

@yuzihe Did you get it through? Looking forward to your update. Thanks!

yuzihe, 2019-05-21:
> 
> 
> @yuzihe Did you get the issue with Debug build or Release build? Could you please try Release build which should be faster than Debug? Thanks!

@RealSenseCustomerSupport  yes,the Release build is faster than Debug, Thanks!

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@yuzihe Thanks for the update! Will close the ticket accordingly.

closed 2019-06-03 by RealSenseCustomerSupport