realsenseai/librealsense #3916: How to get object distances and angles in unity using depth maps?

HCCSY, 2019-05-06:
How to get object distances and angles in unity using depth maps?

HCCSY, 2019-05-06:
How can I use these examples to get the depth and detect the object's Angle after I import realsense.unitypackage in unity?

RealSenseCustomerSupport, 2019-05-14:
----------------------------------------------

Hi @HCCSY,

You can review the script "DepthCutOff.cs" to understand how to get the depth. And, probably, you can evaluate the code of rs-depth-quality tool, which calculate the angle with the target plane.

HCCSY, 2019-05-14:
Thanks for your help! But I really want you to write an example @RealSenseCustomerSupport

RealSenseCustomerSupport, 2019-05-23:
----------------------------------------------

You can try below code to see how it works first.

                var pipe = new Pipeline();
                var config = new Config();
                config.EnableStream(Stream.Color, 1280, 720, Format.Bgr8, 30);
                config.EnableStream(Stream.Depth, 1280, 720, Format.Z16, 30);
                PipelineProfile profile = pipe.Start(config);
                using (var frames = pipe.WaitForFrames())
                using (var depth = frames.First(x => x.Profile.Stream == Stream.Depth) as DepthFrame)
                {
                        Console.WriteLine("The camera is pointing at an object " +
                        depth.GetDistance(depth.Width / 2, depth.Height / 2) + " meters away\t");
                }

RealSenseCustomerSupport, 2019-07-26:
----------------------------------------------

Hi @HCCSY,

Still need the support for this topic? If not, I will close this.

closed 2019-09-25 by RealSenseCustomerSupport