realsenseai/librealsense #3917: Is there are sample python code for D435i getting IMU information?

spk921, 2019-05-06:
Is there are sample python code for D435i getting IMU information?

beduffy, 2019-07-09:
I'm also interested in this for the python wrapper:

`config.enable_stream(rs.stream.pose)`  # crashes here `RuntimeError: Couldn't resolve requests`

if I try to have an empty config I get 
`pose = frames.get_pose_frame()`
pose is a null pointer so later on `pose_frame = np.asanyarray(pose_frame.get_data())` crashes

Similar with accel and gyro. 
`AttributeError: 'pyrealsense2.pyrealsense2.composite_frame' object has no attribute 'get_accel_frame'`

Is C++ the only option for this?

Using realsense 2.23 at the moment since 2.24 has another problem.

beduffy, 2019-07-12:
https://github.com/IntelRealSense/librealsense/issues/2718
https://github.com/IntelRealSense/librealsense/issues/3070
https://github.com/IntelRealSense/librealsense/issues/3101
https://github.com/IntelRealSense/librealsense/pull/3053
https://github.com/GruffyPuffy/imutest
https://software.intel.com/sites/products/realsense/slam/developer_guide.html
https://www.intelrealsense.com/how-to-getting-imu-data-from-d435i-and-t265/

Haven't solved this yet, but I think I will solve it from reading the above. Should be useful for others trying to find out as well. Will update once I work it out.

beduffy, 2019-07-17:
Got the answer from https://github.com/IntelRealSense/librealsense/issues/3409
The API is a bit strange on this one.

```
config.enable_stream(rs.stream.accel, rs.format.motion_xyz32f, 250)
config.enable_stream(rs.stream.gyro, rs.format.motion_xyz32f, 200)
frames = pipeline.wait_for_frames()
for frame in frames:
    motion_data = frame.as_motion_frame().get_motion_data()
    print(motion_data)
    # prints: x: -0.0294199, y: -7.21769, z: -6.41355 for me
    # to get numpy array:
    print(np.array([motion_data.x, motion_data.y, motion_data.z]))

```

gyro is similar. closing.

closed 2019-07-22 by dorodnic