realsenseai/librealsense #3922: Multicam bottlenecks?

atyshka, 2019-05-06:
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400 | 
| Firmware Version                   | Latest | 
| Operating System & Version |  Ubuntu 18.04 (Jetson)| 
| Kernel Version (Linux Only)    | 4.9                                        | 
| Platform                                 | NVIDIA Jetson  |
| SDK Version                            |  2.21.0                          | 
| Language                            |  C++ (ROS)                       | 
| Segment			|  Robot              | 

### Issue Description
We are exploring a 360 degree stereo setup for a robot that would likely involve more than four D435 cameras. We are considering a Jetson Xavier for processing all of this data and I’m trying to assess performance bottlenecks for such a setup. Hopefully USB shouldn’t be a problem, we’re prepared to get an extra pcie usb controller. The Jetson has 8 arm cores and a powerful gpu, so this seems like it should possible. Has anyone looked into profiling the most time-consuming processes for gathering both depth images and rgb? Image format conversion? Stereo matching?

MartyG-RealSense, 2019-05-07:
A good starting point would be pages 5 and 6 of Intel's multiple camera white paper, as these contain tables of estimates for bandwidth usage in various modes for 4 cameras (and some projected values for 5-6 cameras).

https://www.intel.co.uk/content/www/uk/en/support/articles/000028140/emerging-technologies/intel-realsense-technology.html

Advice about latency vs performance is available in the 'Frame Buffering Management' section of the official RealSense documentation.

https://github.com/IntelRealSense/librealsense/wiki/Frame-Buffering-Management-in-RealSense-SDK-2.0

atyshka, 2019-05-07:
Thanks @MartyG-RealSense. I am familiar with that document, it seems that it mostly focuses on bandwidth though and doesn't go too much into processing. Xavier has 2 internal usb controllers and more can be added with pcie, and since we don't need super high res I'm guessing bandwidth shouldn't be an issue. My main concern is processing, especially on an ARM CPU. Do you have any info on a breakdown of the processing (maybe in ms) for a typical rgb-pointcloud pipeline?

MartyG-RealSense, 2019-05-07:
The deeper detail of how a camera image is processed is outside of my knowledge, unfortunately.  An Intel RealSense team member such as @dorodnic or @RealSense-Customer-Engineering may be able to help you with that question.

As the Vision Processor D4 hardware component inside the camera does initial processing on the captured data and therefore helps to avoid putting pressure on a computing device's CPU / GPU, I would speculate that the Jetson Xavier's specification is more likely to have an impact on performance if you are going to be recording the data instead of just live-streaming it (since the speed of the storage medium and how well the device's CPU works with the recording compression algorithm can be factors in performance).

A brochure about the Vision Processor D4 component in the camera may give you useful guidance about its capabilities,

https://www.stemmer-imaging.com/media/uploads/cameras/13/136603-Intel-RealSense-D4-Vision-Processor-Brochure.pdf

atyshka, 2019-05-10:
@MartyG-RealSense  Okay, thanks for the info. I guess I didn't realize stereo processing happened on device. One other question I had was about the ability to dynamically change the camera resolutions. Running many cameras at high resolution might stress the processing/usb bandwidth. A scenario where we could stream all cameras at low res and then get a higher res image on a particular view as needed might be ideal. Currently it seems that the driver must be restarted to change resolutions. @dorodnic Is this a camera firmware-level limitation or a driver limitation?

MartyG-RealSense, 2019-05-10:
You could investigate dynamic reconfigure using ROS, a mode which is activated by starting the stream with a **roslaunch** instruction and then starting the dynamic reconfigure service on the next line with the instruction **rosrun rqt-reconfigure rqt-reconfigure**

https://github.com/IntelRealSense/realsense-ros#set-camera-controls-using-dynamic-reconfigure-params

closed 2019-06-01 by RealSenseCustomerSupport