realsenseai/librealsense #3923: Object move speed

ArtlyStyles, 2019-05-07:
Is there a requirement for object moving speed? I have a robot moving around 10-15 mph, can I use T265 to track the robot position?

bfulkers-i, 2019-05-07:
How well this works will depend very much on the characteristics of the movement (not just speed, but also rotational velocity and accelerations) and the environment the robot is in. The only way to be sure of the tracking quality is to test T265 in your use case. We have seen T265 used successfully in robotics and drone applications with speeds similar to or higher than you suggest.

Also keep in mind that for robotics applications we strongly recommend wheel odometry input to achieve maximum quality.

ArtlyStyles, 2019-05-08:
I race in the DIY Robocars meetup in Oakland. https://www.meetup.com/DIYRobocars/ Currently, my car uses a Lidar and particle filter to locate itself. I am wondering if I can use T265 tracking to replace Lidar.

ArtlyStyles, 2019-05-08:
Thanks. I think I can at least use it to improve my odom, but Lidar and particle filters are still needed.

RealSenseCustomerSupport, 2019-08-02:
----------------------------------------------

Hi @ArtlyStyles,

I am interested if you have any test result with T265?

ArtlyStyles, 2019-08-04:
Yes, please see my twitter https://twitter.com/SmallpixelCar/status/1152765212409229312 In summary, I think the device did a good job and was better than my wheel odom, but was not good enough to replace Lidar and my localization algorithm for my robocar.

RealSenseCustomerSupport, 2020-01-02:
----------------------------------------------

Hi...thanks for providing the feedback. Will you be needing anything else on this particular topic or can we close this?

Also a few updates have been released since August which should improve performance even more if you have not given it a try yet.

MartyG-RealSense, 2020-03-04:
Closed due to lack of further responses after 2 months.

closed 2020-03-04 by MartyG-RealSense