realsenseai/librealsense #3924: Realsense Viewer Crashes with D435/D435i on RPi3

henrikcheung, 2019-05-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435/D435i } | 
| Firmware Version                   | (Realsense 05.10.13.00) | 
| Operating System & Version |   {Linux (Ubuntu 18.04)   | 
| Kernel Version (Linux Only)    |  (4.15.0)                                         | 
| Platform                                 | Raspberry Pi 3+  |

### Issue Description
<Describe your issue / question / feature request / etc..>

~/Desktop$ realsense-viewer

I can open the the realsense viewer, when I start streaming, then it crashes and shows the logs below. 

 07/05 12:41:33,018 WARNING [1841684976] (sensor.cpp:365) Unregistered Media formats : [ P UYVY Y8   ]; Supported: [ GREY P UYVY Y8   Z16  ]
Segmentation fault

I have tried both D435 and D435i and they have the same situation.
How may I fix it please? Thanks!

RealSenseCustomerSupport, 2019-06-06:
----------------------------------------------

@henrikcheung Could you please provide more information as below?
1. Which librealsense version when you got the issue?
2. Was the camera connected to USB2 or USB3 when you get the issue?
3. Which streaming did you enable and the resolution/FPS configurations when get the issue? If there's snapshot for the issue will be also helpful. Thanks!

RealSenseCustomerSupport, 2019-06-13:
----------------------------------------------

@henrikcheung Did you get through the issue? Could you please provide more information if you still need support on this? Thanks!

RealSenseCustomerSupport, 2019-06-17:
----------------------------------------------

@henrikcheung Could you please update?

closed 2019-06-17 by henrikcheung