realsenseai/librealsense #3925: Feature Request: Auto Exposure Maximum Time Limit for IR sensors

mhkabir, 2019-05-07:
When using the D435 IR frames indoors for visual odometry, we have found that the auto-exposure algorithm uses the full available time between frames to expose the IR frames. In general the AE algorithm works quite well, but long exposure times are sub-optimal for use with visual tracking use-cases as it introduces significant motion blur. For these cases, it is far better to use darker images with less motion blur.

Other machine vision cameras (e.g FLIR Blackfly series) provide a feature to limit the maximum exposure time the AE algorithm can command, and I think such an option would be very useful on the Realsense series. It would essentially just cap the commanded shutter time to some configurable maximum (e.g 1-3ms is good for visual odometry) and continue using the sensor's gain functionality once the exposure command is maxed out.

Would be great to hear your thoughts!

mhkabir, 2019-05-07:
@dorodnic @doronhi Your comments would be appreciated! Thanks for the great work so far!

dorodnic, 2019-05-08:
Thanks @mhkabir 

Happy to comment, your request is certainly welcomed and, to me at least, makes sense.

Just to align expectations - I'm managing our open-source software on GitHub and @doronhi maintaining our ROS offering. Neither of us has final say in firmware / hardware issues and requests. 
All GitHub enhancement requests are automatically escalated to relevant hardware / firmware / sensor forums and the outcome is communicated by @RealSenseCustomerSupport 

Did you try increasing the FPS? At second largest resolution the camera can work at up to 90 FPS (not quite 1-3ms you are looking for, but might help a bit)
Perhaps @agrunnet (RealSense CTO) will have an opinion / suggestion on this. 

In addition, while this is not 100% related, for purpose of tracking I wanted to make sure you are aware of [laser on-off toggle](https://github.com/IntelRealSense/librealsense/pull/3066) feature, as well as the more general [exposure, gain and laser toggling](https://github.com/IntelRealSense/librealsense/issues/3844) feature.

mhkabir, 2019-05-09:
Thanks! Looking forward to any comments from @RealSense-Customer-Engineering and @agrunnet.

> In addition, while this is not 100% related, for purpose of tracking I wanted to make sure you are aware of laser on-off toggle feature, as well as the more general exposure, gain and laser toggling feature.

Yes! We use the laser toggling feature in production right now. I'm currently looking into the multi-shot sequencing mode to see if any gains are to be had from it in our use case.

Here is one of our vehicles running a single D435 for visual-inertial odometry and obstacle avoidance.
[image]

[image]

dorodnic, 2019-05-15:
After speaking with @agrunnet we agree it should be added to the pipe. 
@RealSenseCustomerSupport - could you please open a CR?

mhkabir, 2019-05-16:
One more comment : Controlling the maximum command-able auto-gain would also be a very useful feature to control noise in the image, along the same lines of reasoning.

mhkabir, 2019-07-23:
@dorodnic Any updates on this?

RealSenseCustomerSupport, 2020-03-10:
----------------------------------------------

Hi @mhkabir This is in our pipeline. But we don't have a fixed schedule yet.

RealSenseCustomerSupport, 2021-02-09:
----------------------------------------------

Hello,

Just a heads up. The IR camera Max Exposure/Gain Limit feature will be available in the coming RS Viewer 2.42.0 with FW5.12.11.
The attached screenshot shows an example:
Max Exposure=1000 (1000us), NOTE: "0" disable this feature (auto exposure back to normal).
Max Gain=16 (1x gain). "0" disable this feature (auto gain back to normal).

Thanks

Attachment(s):
AE_Max_Limit.JPG - https://realsensesupport.intel.com/attachments/token/S0IWFQJ3Sxm9pqyI3M8qB4LEU/?name=AE_Max_Limit.JPG

closed 2021-02-16 by RealSenseCustomerSupport