realsenseai/librealsense #3926: Error installing librealsense

slucasm, 2019-05-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | R200                                                       | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |  Win 10                                                   | 
| Kernel Version (Linux Only)    |                         | 
| Platform                                 | PC                                                          |
| SDK Version                            |  legacy                     | 
| Language                               |  Python                                                     | 
| Segment			        |  Dron                                                             | 

### Issue Description

I'm trying to use pyrealsense in python for some project with my R200. As the documentation specifies first of all is necessary install librealsense.

To install it I downloaded the folder from branch legacy and build the project in Visual Studio 2015 and run some examples. First of all I detected that the examples only open a terminal and then dissapears. (This is normal?). Once I builded the solution, install depencies as numpy, cython, etc and set enviromental variables PYRS_INCLUDES and PYRS_LIBS I try to install pyrealsense.

When I try to install pyrealsense in the console appears this error:

[image]

I set the variables like this: 

[image]

Thank you for the help in advance.

lramati, 2019-05-19:
As mentioned on the legacy branch, pyrealsense is community maintained. Try asking on their github?
If you're having trouble installing pyrealsense2, then keep in mind it only works with librealsense 2.X.X, but we can help you with any issues you're having

RealSenseCustomerSupport, 2019-06-03:
----------------------------------------------

@slucasm Any other questions about this ticket? Looking forward to your reply. Thanks!

closed 2019-06-13 by RealSenseCustomerSupport