realsenseai/librealsense #3928: Unreliable position output from t265 tracking camera

orshefi, 2019-05-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | t265 | 
| Firmware Version                   | 0.0.18.5129 | 
| Operating System & Version |   Linux (Ubuntu 16)  | 
| Kernel Version (Linux Only)    |  4.15.0-48-generic  | 
| Platform                                 | PC |
| SDK Version                            |  2.21.0 | 
| Language                            |  C / C++ | 
| Segment			|  Robot  | 

### Issue Description
We are trying to use t265 camera to localize our drone in space but we are keep on facing glitches in position.
We built our own handler to the t265 based on the examples given in the sdk.

I will attach rosbags with the camera data while experiancing those glitches, please help us to understand the issue and fix it.
[t265_rosbags.zip](https://github.com/IntelRealSense/librealsense/files/3152886/t265_rosbags.zip)

Filmengineers, 2019-05-12:
Facing same issue here! Its very unreliable. Looking for alternate options. Any suggestions?

bbellhouse, 2019-05-13:
Just as a point of reference, we found that using a Windows Mixed Reality HMD ($300) mounted to our robot and connected to a VR capable machine has provided more consistent inside-out tracking.  Hopefully this problem can be resolved on the Intel t265 tracker since it is based on the same tracking technology as Microsoft's solution with the benefit that it does not require connection to a high-end computer.

RealSenseSupport, 2020-09-09:
Thank you for highlighting the issue with the T265 position tracking output reliability. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

closed 2020-09-16 by RealSenseSupport